RfidResearchGroup/ChameleonUltra #1: Development/testing phase

Kolsha, 2023-02-28:
Hi!
First of all, great and useful project, thanks a lot.
I wonder, can I join development/testing phase?

xianglin1998, 2023-03-01:
We welcome anyone to join the development/testing. The hardware crowdfunding of Chameleon Ultra is about to begin (March). A powerful Chameleon Ultra hardware and your join will make firmware and clients more interesting.

asjdf, 2023-03-01:
Great project!
Is there anything I can do to help? Perhaps issue can be used to manage project progress and let developers know what have to do.

Thanks.

xianglin1998, 2023-03-05:
@Kolsha 

I'm sorry for my late reply, but it is meaningful, because I have had a meeting with Proxgrind's team to discuss it during this time, and I think it is really necessary to arrange the development process clearly, so that fans who like this project can develop and customize according to their own capabilities and needs.

1、 First of all, hardware. All software development should be based on a convenient hardware platform. ChameleonUltra and ChameleonLite are PCBs carefully designed by Proxgrind team. They inherit the style of ChameleonTiny, which is very consistent with Olaf's vision. Yes, whether it is Proxmark3 RDV4 or Chameleon Tiny, Olaf likes to reduce the size as much as possible under the extreme performance, making it easy to carry, unique in appearance and full of technology. At the same time, it is a pity that the volume of Chameleon Ultra brings inconvenience to most people for PCBA manufacturing and firmware development and testing. Therefore, we feel that we need to provide a development board based on ChameleonUltra with a larger volume. The components of this development board will be packaged with common volume to make it easy to weld, and we will also reserve GPIO interface and SWD interface. And this development board is also open source. If enthusiasts are interested, they can build a Chameleon Ultra from scratch. Of course, if there is no welding tool, we also provide formed development boards for sale (moderate price).

2、 Secondly, firmware. Chameleon Ultra is a very powerful platform. We can develop various 14A protocol card simulations based on this platform. In terms of high-frequency card simulation, DXL has completed the development of high-performance MifareClassic Mini ->S50 ->S70 card. However, the development of cards for other technologies has not been completed, such as UL, UL-C, NTAG, etc. In terms of low-frequency card simulation, DXL has completed the development of EM410x, but the development of other technology cards has not been completed, such as FSK and PSK-related technology cards. For card reader, DXL has completed the reading, writing and decryption of M1 card, reading EM410x and writing to T5577. Others, such as the reading and writing of UL card and NTAG card, are not completed yet.

3、 Finally, it is software. At present, ChameleonUltra uses python to develop the CLI, cooperate with independent exe to perform the decryption task, and complete the decoupling of the business logic of the CLI and the decryption algorithm logic. Due to time reasons, DXL only completes commands such as device connection, M1 card data upload, M1 card detection log download and decryption, M1 card read and write block, and so on. The complete nested, rdsc, and autopwn instructions like PM3 have not yet been implemented, which may require our joint efforts. Finally, I believe that Chameleon Ultra is definitely not limited to CLI as a way of device interaction. Perhaps we can further develop PC-side GUI, and develop IOS and Android APP. USB+BLE support makes everything possible.

ShigemoriHakura, 2023-03-05:
I found that the sch files seems mismatch with the Gerber file, the HF switch seems not 1G3157.

mitmarcus, 2023-08-29:
Is there an explanation on the LED color? 

green/red/blue

doegox, 2023-09-27:
Notes transferred into https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/technical_whitepaper.md#birth-of-the-project

closed 2023-09-27 by doegox