RfidResearchGroup/ChameleonUltra #8: Attack Information needed!

derGraph, 2023-05-05:
Where can I find any information about the attacks, like how do they work on an firmware level.

Thanks

GameTec-live, 2023-05-06:
reverse engineer the code, afaik they are performed on the PC:
https://github.com/RfidResearchGroup/ChameleonUltra/tree/main/software/src
https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/software/script/chameleon_cmd.py

derGraph, 2023-05-06:
I can read code, but I woud like to understand the code im reading!

GameTec-live, 2023-05-06:
cant help you on that, sorry... (google ig?)

xianglin1998, 2023-05-11:
# Such as mifareclassic attack:
1、Attack at chameleon firmware side(nonces acquire):
[Acquire](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/firmware/application/src/rfid/reader/hf/mf1_toolbox.c)
2、Attack at pc side(nonces decrypt):
[Decrypt Algorithm Implement](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/software/src)
[Decrypt Task Invoke](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/software/script/chameleon_cli_unit.py)

xianglin1998, 2023-05-11:
If you want to try to understand its principles, you may need to purchase a Proxmark3 RDV4 to learn RFID knowledge. Believe me, this is a very difficult and interesting process.

closed 2023-09-27 by doegox