RfidResearchGroup/ChameleonUltra #9: How to open project!

derGraph, 2023-05-07:
I installed the SEGGER embedded Studio, but how do I open the firmware?

Thanks

GameTec-live, 2023-05-08:
maybe that will help? 
(altough unsure if they even use segger studio?)
https://www.segger.com/downloads/embedded-studio/EmbeddedStudio_Manual

xianglin1998, 2023-05-11:
I'm use vscode + armgcc + openocd to develop.

xianglin1998, 2023-05-11:
[image]

xianglin1998, 2023-05-11:
I will find time later to write the deployment tutorial for the development environment into the markdown, and I have been busy dealing with the production of the chameleon Ultra recently.

derGraph, 2023-05-21:
I was able to open and compile the project! I wrote some instructions to the 
Did you use the VS-Code Cortex-Debug Extension? What version of GCC did you use? Can you share your openocd commands/files?
How did you compile the bootloader and application in one file or did you upload them seperate?`
Can you share your flashing procedure?

thanks,
Paul

xianglin1998, 2023-05-22:
Good~~!!!
### A、Yes, i'm using cortex-debug extension on vs, and jlink gdb tools(not openocd)
this is content for my launch.json file
```
{
    "configurations": [
        {
            "cwd": "${workspaceFolder}",
            "executable": "${workspaceRoot}/firmware/objects/bootloader.out",
            "name": "Debug with JLink",
            "request": "launch",
            "type": "cortex-debug",
            "runToEntryPoint": "main",
            "showDevDebugOutput": "none",
            "servertype": "jlink",
            "device": "nrf52",
            "interface": "swd",
            "svdFile": "${workspaceRoot}/firmware/nrf52_sdk/modules/nrfx/mdk/nrf52.svd"
        },
    ]
}
```
Edit your cortex-debug settings in vs settings.json, to define
```
    "cortex-debug.armToolchainPath": "Your arm gcc toolchain path",       // such as C:\\UserProgram\\arm_gcc\\none\\bin
    "cortex-debug.JLinkGDBServerPath": "Your jlink gdb executable path",  // such as C:\\Program Files (x86)\\SEGGER\\JLink\\JLinkGDBServerCL.exe
```
Note: you can replace bootloader.out to application.out to debug app.

### B、I'm use armgcc-none to compile firmware for chameleonUltra, download from Official  website
[Download from Official  Website For ARM](https://developer.arm.com/downloads/-/gnu-rm)
and it's version:
[image]

### C、Use "nrfutil" merge boot and app firmware hex to one file(To install tool from NRF website)
```
// generate settings page, get a 'settings.hex'
nrfutil settings generate --family NRF52840 --application application.hex --application-version 1 --bootloader-version 1 --bl-settings-version 2 settings.hex
// merge boot and settings page to one file, get a 'bootloader_settings.hex'
mergehex --merge bootloader.hex settings.hex --output bootloader_settings.hex
// merge boot settings and app to one file, get a 'project.hex'
mergehex --merge bootloader_settings.hex application.hex softdevice.hex --output project.hex
```
Note: You can flash only application firmware to debug, or else the bootlaoder has crc check, your new app from jlink flash will error start and goto bootloader.

### D、Flash firmware and Enjoy your dev.
0、Open terminal in projectRoot/firmware/[application ] or [bootloader]
1、Execute cmd `make -j4` to compile [application ] or [bootloader]
2、Execute cmd `make flash` to flash firmware(only one, if your device is new chip, you must to flash soft device!!!)
3、Execute cmd `make flash_softdevice` to flash softdevice(Optional~~~)
Note: the make flash cmd will auto execute `nrfjprog xxx xxxx`, If you are interested, you can read the official NRF documentation.
[image]

xianglin1998, 2023-05-22:
@derGraph

derGraph, 2023-05-22:
Thanks, I will close this issue when I can get it to run + written instructions!

xianglin1998, 2023-05-23:
> Where did you get the softdevice.hex file?

In the {projectRoot}/nrf52_sdk/components/softdevice/s140/hex/s140_nrf52_7.2.0_softdevice.hex

derGraph, 2023-06-10:
Everything clear and working, if possible please check my instructions for JLink (I have only STLINK)
Instructions in new Pull request #14 !

closed 2023-06-10 by derGraph