RfidResearchGroup/ChameleonUltra #11: Consider adding a license

GameTec-live, 2023-05-25:
Consider adding a license to this project, in my opinion a GPL V3 or something similar would fit just fine.

(How do i come to GPL V3? a) I like the GPL V3 b) the Proxmark and some other repos are already licenes under GPL V3, so why not this one?)

xianglin1998, 2023-05-29:
The Chameleon Ultra uses some code from Proxmark3 RRG Repo, perhaps we should continue the license statement from Proxmark3 RRG Repo.

GameTec-live, 2023-05-29:
wouldnt the gpl v3 (under which the pm3 repo is licensed?) require the licensing under gplv3 if parts of that code is used?
also: does indeed make sense

xianglin1998, 2023-05-29:
I am not familiar with GPL V3, which may require ICEMAN's help.

GameTec-live, 2023-05-29:
from my understanding, if you use part of GPL licensed code, the project this licensed code is in must also be licensed under the same GPL version? (eg GPL V3)

GameTec-live, 2023-05-29:
[image]
github even has a neat summery XD

xianglin1998, 2023-05-30:
Yes, GPLV3 is inheritance. Our project should be GPLV3 or more strictly licensed, but I think GPLV3 should be enough. A good project cannot be separated from more people/groups that create value for themselves or others, sure, must following GPLV3.

xianglin1998, 2023-05-30:
The project start follow GPLV3.
Look commit: a048187

closed 2023-05-30 by xianglin1998