RfidResearchGroup/ChameleonUltra #16: Notes about compiling and running under Linux

doegox, 2023-07-01:
I'm sharing here personal notes, in the hope it can help other people willing to contribute to development from a Linux environment (I'm using a Debian testing).
Things are still a big foggy so I prefer to not merge what is working for me in the official doc yet.
Thanks to the early hackers who shared useful findings not yet merged at the time of writing ( https://github.com/RfidResearchGroup/ChameleonUltra/pull/15/files and https://github.com/RfidResearchGroup/ChameleonUltra/pull/14/files)

# Compiling FW

Using Debian `gcc-arm-none-eabi` 12.2.1, the compilation halts on a few warnings that must be fixed.
Meanwhile, we can force compilation by adding to `firmware/application/Makefile` and `firmware/bootloader/Makefile`: `CFLAGS+=-Wno-error=array-bounds -Wno-error=sizeof-array-div`

As we're using the system ARM cross-compiler, we can configure `firmware/Makefile.defs` in a very generic manner:

```
GNU_INSTALL_ROOT ?= 
GNU_PREFIX ?= arm-none-eabi
```
This could become the default choice in the repo code...

One last thing, with ARM GCC 12.2.1 I got an error about the bootloader to be 128b too large. So I changed the boundaries as follows.

`firmware/application/application.ld`:
```diff
-  FLASH (rx) : ORIGIN = 0x27000, LENGTH = 0xCC000
+  FLASH (rx) : ORIGIN = 0x27000, LENGTH = 0xCB000
```
`firmware/bootloader/bootloader.ld`:
```diff
-  FLASH (rx) : ORIGIN = 0xF3000, LENGTH = 0xB000
+  FLASH (rx) : ORIGIN = 0xF2000, LENGTH = 0xC000
```
Now we can compile the firmware:
```sh
(
  cd firmware
  rm -rf objects
)
(
  cd firmware/bootloader
  make
)

(
  cd firmware/application
  make
)

(
  cd firmware
  ../../tools/nrfutil settings generate --family NRF52840 --application objects/application.hex --application-version 1 --bootloader-version 1 --bl-settings-version 2 objects/settings.hex
  mergehex --merge objects/bootloader.hex objects/settings.hex --output objects/bootloader_settings.hex
  mergehex --merge objects/bootloader_settings.hex objects/application.hex nrf52_sdk/components/softdevice/s140/hex/s140_nrf52_7.2.0_softdevice.hex --output objects/project.hex
)
```
The final image is `firmware/objects/project.hex`

# Flashing FW

## With stlink v2:

Connect
* Ultra <> Stlink V2 (translucent green)
* Vcc left disconnected
* SWC <> 4:SWCLK
* SWD <> 6:SWDIO
* GND <> 7:GND
* SWO left disconnected

Do not connect over USB at same time, it will work over battery.

```sh
openocd -f interface/stlink.cfg -f target/nrf52.cfg -c "flash init; init; reset halt; flash erase_sector 0 1 last; exit"
openocd -f interface/stlink.cfg -f target/nrf52.cfg -c "program firmware/objects/project.hex verify reset ; shutdown"
```

## With jlink:

Connect
* Ultra <> Jlink
* Vcc left disconnected
* SWC <> 9:SWCLK
* SWD <> 7:SWDIO
* GND <> 8:GND
* SWO left disconnected

```sh
openocd -f interface/jlink.cfg -c "transport select swd" -f target/nrf52.cfg -c "flash init; init; reset halt; flash erase_sector 0 1 last; exit"
openocd -f interface/jlink.cfg -c "transport select swd" -f target/nrf52.cfg -c "program firmware/objects/project.hex verify reset ; shutdown"
```

# Python CLI

## Setup

```sh
(
  cd software/script
  virtualenv venv
  source venv/bin/activate
  pip install -r requirements.txt
  deactivate
)
```

## Usage

```sh
(
  cd software/script
  source venv/bin/activate
  python chameleon_cli_main.py
  deactivate
)
```
To connect once in the CLI:
```
hw connect -p /dev/ttyACM0
```

# Software (unused so far)

## Setup

```
(
  cd software/src
  rm -rf tmp
  mkdir tmp
  cd tmp
  cmake ..
  make
  echo "DONE. bins in software/bin"
)
```

xianglin1998, 2023-07-03:
I am using ARM GCC version 9.3.1, and the link script works normally on this version of the compiler without reporting the error 'bootloader to be 128b too large'. At present, the ROM boundaries of the bootloader and application for mass production have been confirmed. If this boundary is modified, will it cause the main program to freeze after subsequent OTA, or will there be issues with FDS(flash r/w library) exceeding the boundaries?

xianglin1998, 2023-07-04:
[image]
I downloaded an armgcc test compilation for version 12.2.1 and found that it can pass (without changing the link script). But I am cross compiling under Windows, and I am not sure if it will affect the test results.

doegox, 2023-07-11:
Hmm strange. We might try to compare build maps but I guess we've little control about the size of the bootloader created by a given toolchain.
Is it acceptable to leave the bootloader start image where it is but let it go beyond 0x000FFFFF ?
I tried the following and reflashed the dev board over DFU without any issue:
```diff
+++ b/firmware/bootloader/bootloader.ld
@@ -5,12 +5,12 @@ GROUP(-lgcc -lc -lnosys)
 
 MEMORY
 {
-  FLASH (rx) : ORIGIN = 0xF3000, LENGTH = 0xB000
+  FLASH (rx) : ORIGIN = 0xF3000, LENGTH = 0xB800
   RAM (rwx) :  ORIGIN = 0x20005978, LENGTH = 0x32688
   uicr_bootloader_start_address (r) : ORIGIN = 0x10001014, LENGTH = 0x4
-  bootloader_settings_page (r) : ORIGIN = 0x000FF000, LENGTH = 0x1000
+  bootloader_settings_page (r) : ORIGIN = 0x000FF800, LENGTH = 0x1000
   uicr_mbr_params_page (r) : ORIGIN = 0x10001018, LENGTH = 0x4
-  mbr_params_page (r) : ORIGIN = 0x000FE000, LENGTH = 0x1000
+  mbr_params_page (r) : ORIGIN = 0x000FE800, LENGTH = 0x1000
 }
```

xianglin1998, 2023-07-12:
[bootloader.zip](https://github.com/RfidResearchGroup/ChameleonUltra/files/12023478/bootloader.zip)
This is my compile output map of bootloader.

nvx, 2023-07-13:
@doegox I ran into the same issue on Windows, turns out the arm-none-eabi-gcc as packaged by MSYS2 seems to produce a larger sized bootloader than the pre-built binary toolchain as downloaded from https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads

I'm guessing there's some tweaks to the gcc source accounting for the differences.

doegox, 2023-07-13:
and on the opposite side, @augustozanellato could compile fine under linux, but using the https://armkeil.blob.core.windows.net/developer/Files/downloads/gnu/12.2.rel1/binrel/arm-gnu-toolchain-12.2.rel1-x86_64-arm-none-eabi.tar.xz  (cf https://github.com/augustozanellato/ChameleonUltra/blob/main/firmware/Dockerfile )
So it's not a Windows/Linux issue it's a upstream<>Debian issue.

xianglin1998, 2023-07-18:
This is too interesting, so can we assume that the firmware is okay?

xianglin1998, 2023-07-18:
I dare not change the range of flash at will because I am afraid of space overflow leading to strange bugs. And I calculated the control range according to the official manual.

**NRF52840 Memory Map**

> [Doc Link](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v12.2.0%2Flib_bootloader_dfu_process.html)

[image]

doegox, 2023-07-26:
I'm closing it now, info got consolidated and merged in https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/How_to_use_Firmware.md
thank you all for the feedbacks!

xianglin1998, 2023-07-27:
Nice END.

closed 2023-07-27 by doegox