RfidResearchGroup/ChameleonUltra #20: Port autodetect does not work on WSL1

gentilkiwi, 2023-07-11:
https://github.com/RfidResearchGroup/ChameleonUltra/blob/939df3f0aa9525d9754a011228cc6c0bda087d8b/software/script/chameleon_cli_unit.py#L200

`Chameleon Connect fail: 'NoneType' object has no attribute 'startswith'`

Not working on `WSL1`, it seems @iceman1001 use a PowerShell hack to detect it under Windows env. then pass results to scripts

xianglin1998, 2023-07-12:
This looks like there's no manufacturer information available

xianglin1998, 2023-07-12:
I have installed wsl&&ubuntu and used usbipd for USB device forwarding. Currently, my test results are normal, and I suspect that you cannot recognize the manufacturer information because the driver is not installed correctly, resulting in incorrect acquisition of the manufacturer information or description information.

xianglin1998, 2023-07-12:
[image]
You can use `lsusb` to list all devices and check if chameleonultra is loaded correctly.

gentilkiwi, 2023-07-12:
In `WSL1`, serial ports are natively redirected/exposed

GameTec-live, 2023-07-12:
> In `WSL1`, serial ports are natively redirected/exposed

so, ofc this is still an issue and ig its also partly mine, because i tried implementing that autodetect, but why not use wsl 2? isnt wsl 1 now not even supported anymore? also; compiling and running in proxspace should work too ;)

gentilkiwi, 2023-07-12:
I don't blame anyone @GameTec-live ;)

At the moment it's still supported: https://devblogs.microsoft.com/commandline/wsl-2-post-build-faq/#what-will-happen-to-wsl-1-will-it-be-abandoned
And why use it ? Especially because of serial native supports, it works out of the box without latencies!

Foxushka, 2023-07-13:
> but why not use wsl 2

Because minimum required windows for wsl2 is 20xx. Windows 10 LTSC 2019 based on 18xx. Newer versions of windows are just unusable buggy shit

GameTec-live, 2023-07-17:
its very likely my fault, autodetect doesnt even work in normal windows, it should be fixed with pr #26

gentilkiwi, 2023-07-17:
It does not work with WSL1
> it seems @iceman1001 use a PowerShell hack to detect it under Windows env. then pass results to scripts

xianglin1998, 2023-07-18:
> 

A recent PR #26 has attempted to solve this problem, you can give it a try. ^.^

GameTec-live, 2023-07-18:
> > 
> 
> A recent PR #26 has attempted to solve this problem, you can give it a try. ^.^

if wsl1 doesnt expose anything (as shown above), no vid, pid, etc, i think its not possible to identify a chameleon?

nieldk, 2023-07-22:
You probably need to use usbipd

https://github.com/dorssel/usbipd-win

GameTec-live, 2023-07-22:
> 

they are on wsl one, not 2. cool idea though

Msprg, 2023-08-13:
> > > 
> > 
> > 
> > A recent PR #26 has attempted to solve this problem, you can give it a try. ^.^
> 
> if wsl1 doesnt expose anything (as shown above), no vid, pid, etc, i think its not possible to identify a chameleon?

Perhaps take a peek at how RRG/pm3 client does it. Seems to work reliably on WSL1 at least on my setup.

GameTec-live, 2023-08-13:
> Perhaps take a peek at how RRG/pm3 client does it. Seems to work reliably on WSL1 at least on my setup.

problem being: pm3 is in c, this is python

m-kozlowski, 2023-08-13:
https://github.com/RfidResearchGroup/proxmark3/blob/81a0739153aa2dce7af05e273382b889b777f0c3/pm3#L182

xianglin1998, 2023-08-22:
@gentilkiwi Now have a pr to resolve your problem, can you test it and feedback results, thanks.

PR #73 (2023-08-22): Add port auto detection on WSL1
fixes #20
closed 2023-08-22 by doegox