RfidResearchGroup/ChameleonUltra #29: Can't make flash or make flash_softdevice

retX0, 2023-07-19:
When I running those command, it raise error
 `ERROR: No debuggers were discovered.`
 I'm using the mac mini with m1 chip. and I use command 'JLinkExe -device nRF52 -if SWD' tried to find the device. but failed either. How should I solve this problem or how should i update the bootloader and application?

xianglin1998, 2023-07-21:
Please confirm your device type and connect it to the corresponding debugging probe pin according to your device type.

Note: After erasing all firmware, ChameleinLite needs to use 3.3V power supply for the VCCNRF pin, otherwise it will result in the kernel not having 3.3V power supply and unable to start.

retX0, 2023-07-21:
Hold on. I'm using the type-c cable with ultra, do you mean I need some specific cable to flash the device ?

xianglin1998, 2023-07-21:
If you are using type-c,  Only run usb-dfu can to update your device firmware.

xianglin1998, 2023-07-21:
after `nrfutil ` installed, you can run cmd:
`nrfutil pkg generate --application application.hex --application-version 1 --hw-version 0 --sd-req 0x100 --key-file ../ChameleonUltra/resource/dfu_key/chameleon.pem chameleon_ultra_app_update.zip`
to build update packages.

The above example is to build an update package for an application. If you need to update the bootloader, you need to read the command line documentation for `nrutil`. I have not tried using USB-SERIAL to update the BOOT firmware yet.

This obviously does not apply to downloading firmware multiple times during debugging (which is too troublesome). If you need to modify the functionality/debugging, I suggest you purchase the JLink probe tool.

retX0, 2023-07-23:
Follow your advice, I generated the `app_update.zip` and enter the dfu mode, use command 
`nrfutil dfu serial -b 115200 -p /dev/tty.usbmodemEF38944C806E1 -pkg chameleon_ultra_app_update.zip`
updated the device.

closed 2023-07-23 by retX0