RfidResearchGroup/ChameleonUltra #44: Where is the Documentation / Basic User Guide

GarethWright, 2023-07-27:
Forgive me if I've overlooked something but there's no user guide or basic instructions on how to use it?
Not even a description of what each button does?

GameTec-live, 2023-07-27:
nope, none yet... for the buttons afaik theres some info in the indigogo description.
Proper documentation is coming SoonTM

SieGoz, 2023-07-28:
Hello,
I have the same problem...

GameTec-live, 2023-07-28:
to load something into the slot you need to use client software currently (please be pantient and understand that a lot of this is community driven and also the firmware as well as the client software are in early stages)...

For the client software there are more or less 2 options. The official CLI which supports all currently available functions (It can be found in this repo under the software folder and requires python, etc. watch icemans video on a good tutorial to building it) and my GUI. it works on android windows as well as linux but for example slot managment is still being worked on.

SieGoz, 2023-07-28:
Ok thanks !

pl4nty, 2023-07-29:
These commands worked for me in WSL2 Ubuntu, based on [Iceman's video](https://youtu.be/VGpAeitNXH0). If you're also using WSL2, make sure to [attach the USB device](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) and [set permissions](https://github.com/dorssel/usbipd-win/wiki/WSL-support#udev) first.

Setup:
```
git clone https://github.com/RfidResearchGroup/ChameleonUltra
cd ChameleonUltra/software/src
mkdir build
cd build
cmake ..
cd ../../script
python3 -m pip install -r requirements.txt
```

Usage:
```
python3 chameleon_cli_main.py
hw connect
hw mode set -m e
hw slot openall
```

Select a slot with the physical buttons

```
hf mf eload -t bin -f yourbin.bin
hf mf sim --sak [hex SAK] --atqa [hex ATQA] --uid [hex UID]
```

Can also use `hf mf eload -t hex -f yourhex.eml

GameTec-live, 2023-07-29:
> For the CLI, the following worked for me in WSL2 Ubuntu based on [Iceman's video](https://youtu.be/VGpAeitNXH0). If you're also using WSL2, make sure to [attach the USB device](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) and [set permissions](https://github.com/dorssel/usbipd-win/wiki/WSL-support#udev) first.
> 
> Setup:
> ```
> git clone https://github.com/RfidResearchGroup/ChameleonUltra
> cd ChameleonUltra/software/src
> mkdir build
> cd build
> cmake ..
> cd ../../script
> python3 -m pip install -r requirements.txt
> ```
> 
> Usage:
> ```
> python3 chameleon_cli_main.py
> hw connect
> hw mode set -m e
> hw slot openall
> ```
> 
> Select a slot with the physical buttons. A red LED will indicate the selected slot.
> 
> ```
> hf mf eload -t bin -f yourbin.bin
> hf mf sim --sak [hex SAK] --atqa [hex ATQA] --uid [hex UID]
> ```
> 
> You can also use `hf mf eload -t hex -f yourhex.eml` for hex-encoded dumps.
> 
> Now you can disconnect and use buttons to select the slot. The LED will be red when selected, and green after it's read.

on the latest cli specifying the type wheb using eload isnt required. also: its strongly recommended to update the fw...

you can either use the gui for that or do hw dfu and thn use nrfutil: 
```
nrfutil device program --firmware lite-dfu-app.zip --traitd nordicDfu
```

GibbilyGooo, 2023-07-29:
I’m getting errors at the cmake .. command when using git bash. What do you suggest is the appropriate install for cmake?

Thank you

pl4nty, 2023-07-29:
@GameTec-live thanks for the eload tip, I've updated my comment. how is the firmware versioned? I'm running v1.0 at the moment, but the other releases seem to be rolling. not sure I want to run bleeding-edge firmware at the moment

GameTec-live, 2023-07-29:
> @GameTec-live thanks for the eload tip, I've updated my comment. how is the firmware versioned? I'm running v1.0 at the moment, but the other releases seem to be rolling. not sure I want to run rolling-release firmware at the moment

they arent versioned, because as you guessed, they are rolling... the factory fw is just old and has some bugs (same for v1 release)
good news though, the chameleon is unbrickable, because unless you intentionally unlock it, the bootloader is read only...

GameTec-live, 2023-07-29:
> I’m getting errors at the cmake .. command when using git bash. What do you suggest is the appropriate install for cmake?
> 
> Thank you
> 
> Sent from Proton Mail for iOS
> 
> 
> >> For the CLI, the following worked for me in WSL2 Ubuntu based on [Iceman's video](https://youtu.be/VGpAeitNXH0). If you're also using WSL2, make sure to [attach the USB device](https://learn.microsoft.com/en-us/windows/wsl/connect-usb) and [set permissions](https://github.com/dorssel/usbipd-win/wiki/WSL-support#udev) first.
> >>
> >> Setup:
> >>
> >> git clone https://github.com/RfidResearchGroup/ChameleonUltra
> >> cd ChameleonUltra/software/src
> >> mkdir build
> >> cd build
> >> cmake ..
> >> cd ../../script
> >> python3 -m pip install -r requirements.txt
> >>
> >> Usage:
> >>
> >> python3 chameleon_cli_main.py
> >> hw connect
> >> hw mode set -m e
> >> hw slot openall
> >>
> >> Select a slot with the physical buttons. A red LED will indicate the selected slot.
> >>
> >> hf mf eload -t bin -f yourbin.bin
> >> hf mf sim --sak [hex SAK] --atqa [hex ATQA] --uid [hex UID]
> >>
> >> You can also use hf mf eload -t hex -f yourhex.eml for hex-encoded dumps.
> >>
> >> Now you can disconnect and use buttons to select the slot. The LED will be red when selected, and green after it's read.
> >
> > on the latest cli specifying the type wheb using eload isnt required. also: its strongly recommended to update the fw...
> >
> > you can either use the gui for that or do hw dfu and thn use nrfutil:
> >
> > nrfutil device program --firmware lite-dfu-app.zip --traitd nordicDfu

i recommend you use proxspace or wsl (as shown in video), proxspace has all tools already setup

konkeydong75, 2023-08-01:
Can the device read/emulate Paxton Net2 cards?

GarethWright, 2023-08-01:
HW yes, Firmware not implemented yet.

konkeydong75, 2023-08-01:
Perfect, many thanks @GarethWright

atkinchris, 2023-08-08:
> cmake ..

Depending on your setup, you might need to run `make` after running `cmake ..`, to actually build the binaries for `nested`, etc.

ComicBit, 2023-08-21:
I believe that robust documentation is crucial for any product, especially for newcomers venturing into this domain for the first time. Currently, the absence of thorough documentation starkly differentiates this product from the Flipper Zero. The Flipper Zero stands out not just because of its detailed introduction but also because of its supportive community. If this project continues on its current path, it risks never cultivating such a community. It's disappointing to see such a fundamental aspect being neglected.

ComicBit, 2023-08-21:
Solid documentation is super important for any product, especially for the new folks trying to get into this. Without it, they might just get turned off. Right now, the product's documentation isn't even close to what Flipper Zero has. The Flipper is killing it with both their detailed guides and a kickass community. We're really missing out by not focusing on this. It's kinda frustrating to see honestly.

doegox, 2023-08-21:
Agreed. But hard to document an early moving target with new or changed features every day. And a few million dollars difference. Will you be part of the kickass community and help documenting?

ComicBit, 2023-08-21:
I feel like the project's at a point where we could at least start drafting a basic guide on how to use it. A few examples and maybe some guidance on expanding its capabilities would be great. Creating "app-like" modules for people that want to develop new attacks sounds cool, though I get we're not there just yet. The main aim should be to encourage more contributions. I'm up for helping with docs and building the community, but I need a clearer grasp on the tool. Right now, I'm struggling in the most basic things. About the budget, I'm with you. But if the Chameleon Ultra can outdo what Flipper can't because of its hardware, we might be onto something big and some viral video could bring also some cash. It'd be awesome to see this become the new standard in NFC/RFID testing.

xianglin1998, 2023-08-22:
This is indeed our negligence. Starting from next week, I will try my best to arrange time to complete some of the community's existing needs, and write product's documentation.

nuvious, 2023-08-23:
Let me know if you need help on this issue. I just got mine today and am loving it. Happy to draft some `Quickstart` user guides.

ComicBit, 2023-08-24:
I think we also need as part of this wiki how the device operates, which buttons to press and what does what, what's the bootloader mode etc. I was able to find for instance a reference on the fact that if you don't press the B button (i hope I remember that right) while pairing bluetooth, the device will refuse connection. I could find this only in some platform to purchase the device into the description

GameTec-live, 2023-08-24:
@ComicBit My plan is to put all of that into the FW section 🤷

PR #89 (2023-08-25): Adding a (not so) basic users guide
Im doing it, im adding a users guide to ideally adress as much as possible. (closes #44 )
And because apperently not a lot of people know how rfid works, etc, @nemanjan00 had the idea of creating a RFID wiki (unrelated, but worth mentioning)
closed 2023-08-25 by doegox