RfidResearchGroup/ChameleonUltra #46: Chameleonultra - Device fail to connect <HELP>

sumiej, 2023-08-01:
I am the fresh man to use the chameleonultra, I was download the windows application from Gametech-Live/ChameleonUltraGUI.
 
I was try to activate the developer mode, and use the function " DFU Flash ultra FW".
Device was automatical re-connect and the device is nore more available in GUI.
 
I was check the WIndows device manager, the ChameleonUltra has become the unknow device of USB(as below snap)
 
Would you please help me to assist to advice to recovery the device.
 
Very thanks for you assistance.

[image]
[image]

GameTec-live, 2023-08-01:
sooo....
1) The dev menu is for people who know what they are doing, aka the developerd of the app, the flash dfu your supposed to use is in the settings / cog menu on the homepage right after you connect

2) bricking a chameleon is very hard, either see if it shows up in the gui in dfu mode (make sure your on the latest version, windows dfu only got fixed recently) or use nrfutil as shown in other issues (i think its the getting started documentation issue)

sumiej, 2023-08-03:
thanks I will try.

pmkid44, 2023-08-06:
I have a DFU-mode issue. My chameleon got stuck in DFU mode, when I connect it to the GUI it gets recognized, I choose the flash option at prompt and the device starts showing very fast alternating blue LEDs but nothing happens for more than 15 minutes, there is no progress indication nor activity to be seen on the system. I'm using a mac by the way. Is there any way to get the device out of DFU mode?

GameTec-live, 2023-08-07:
> I have a DFU-mode issue. My chameleon got stuck in DFU mode, when I connect it to the GUI it gets recognized, I choose the flash option at prompt and the device starts showing very fast alternating [...]

the latest gui version should have working dfu, so please make sure your on latest.
 The device _should_ exit dfu automaticslly after ~1min of no activity...

GameTec-live, 2023-08-07:
btw, for GUI this is the wromg issue tracker :p

doegox, 2023-09-27:
is this issue still relevant ?

pmkid44, 2023-09-27:
Mine (in the wrong thread) isn’t relevant anymore. The issues have been solved with a later version rebuild of the GUI app.  Was able to reflash firmware on the first try. Thanks!

closed 2023-09-27 by doegox