RfidResearchGroup/ChameleonUltra #48: Checking firmware version?

grymoire, 2023-08-08:
It would be nice to have a feature where we can check the current firmware version number with the latest version of the firmware. This way those who are having trouble compiling the firmware can ensure they have the latest version installed.

brianbgarber, 2023-08-09:
I was thinking the same thing. I uploaded the latest release of ultra-dfu-app.zip via the chameleonultragui app, but it still shows Firmware Version 1.0.

GameTec-live, 2023-08-09:
tha automatic firmware versioning was removed because... idk... But it was removed for some reason, so currently all builds are 1.0, yes...

grymoire, 2023-08-09:
So there is no way to tell if you are up to date or not? 
May I suggest as an enhancement you offer pre-built files with revision numbers (and dates).

augustozanellato, 2023-08-10:
> 

> tha automatic firmware versioning was removed because... idk... But it was removed for some reason, so currently all builds are 1.0, yes...

It wasn't removed, it just have never been implemented

GameTec-live, 2023-08-10:
> > tha automatic firmware versioning was removed because... idk... But it was removed for some reason, so currently all builds are 1.0, yes...
> 
> It wasn't removed, it just have never been implemented

there was a commit with remove auto fw versioning or smthng...

closed 2023-08-17 by doegox