RfidResearchGroup/ChameleonUltra #55: proxmark error to write  MIFARE Classic 1K to ChameleonUltra

randomk, 2023-08-12:
[image]

Can't set magic card block: 0

xianglin1998, 2023-08-12:
Default mode is Standard mifareclassic tag, cload can't run on this mode, plz use Chameleon Ultra CLI, run `hf mf eload` to load your dump

randomk, 2023-08-17:
i want to record MIFARE Classic 1K card in ChameleonUltra using proxmark

xianglin1998, 2023-08-21:
You should to using `fchk` to check ChamaleonUltra keys and using `restore` to write data for ChameleonUltra by Proxmark3.
[image]

xianglin1998, 2023-08-21:
ChameleonUltra not a `Chinese Magic Card` at default config. You cant write any block using backdoor cmd(Gen1A namely Chinese Magic Card). So `cload` cmd is unavailable.

closed 2023-08-21 by xianglin1998