RfidResearchGroup/ChameleonUltra #58: TimeoutError: CMD 1002 exec timeout - device timeout?

boby498800, 2023-08-16:
Hi, I used the code from the following video to run my ultra chameleon: https://www.youtube.com/watch?v=VGpAeitNXH0
It worked perfectly, however, after playing around with the MTools Lite with my chameleon ultra, I tried using the same commands as the video: hw mode get -h or set and I now get the following errors: 
CLI exception: Traceback (most recent call last):
  File "chameleon_cli_main.py", line 219, in startCLI
    unit.on_exec(args_parse_result)
  File "/Users/XXXX/ChameleonUltra/software/script/chameleon_cli_unit.py", line 236, in on_exec
    print(f"- Device Mode ( Tag {'Reader' if self.cmd_standard.is_reader_device_mode() else 'Emulator'} )")
  File "/Users/XXX/ChameleonUltra/software/script/chameleon_cmd.py", line 118, in is_reader_device_mode
    resp = self.device.send_cmd_sync(DATA_CMD_GET_DEVICE_MODE, 0x00, None)
  File "/Users/XXX/ChameleonUltra/software/script/chameleon_com.py", line 345, in send_cmd_sync
    raise TimeoutError(f"CMD {cmd} exec timeout")
TimeoutError: CMD 1002 exec timeout

Has this happened to anyone else? 

Thanks for your help.

F9Alejandro, 2023-08-17:
Are you attempting to use the CU with bluetooth? It is recommended to use it via cable. Also be sure to disconnect your CU device from mtools lite using the about menu and clicking on the red power logo.

doegox, 2023-09-27:
is it still an issue with the latest fw ?