RfidResearchGroup/ChameleonUltra #62: When will the MIFARE Classic card number be copied offline?

fd099, 2023-08-18:
For example, click the A button to copy the card number。Thanks！

GameTec-live, 2023-08-18:
probably not in the next time, because youd need to recovery keys and the nrf processor is just not powerfull enough...
Altough, yes, just copying UID might be possible...

Markus-HL, 2023-08-18:
only copying UID would be perfect for me. :)

fd099, 2023-08-21:
> probably not in the next time, because youd need to recovery keys and the nrf processor is just not powerfull enough... Altough, yes, just copying UID might be possible...

The current stage only needs to copy UID would be perfect for me

xianglin1998, 2023-08-22:
It's very easy to just copy the UID, stay tuned, it won't be long.

xianglin1998, 2023-08-22:
Implemented At: 7355d512dd2ae480f6a94ca109f4721f91745a94

closed 2023-08-22 by xianglin1998