RfidResearchGroup/ChameleonUltra #90: TypeError: 'type' object is not subscriptable

szikora, 2023-08-24:
I'm new to this project. So, maybe, I miss something...

On Debian Buster 10.13 amd64

After installing the code with the last commit (5c62739) and trying to run python3 chameleon_cli_main.py, I get this:
Traceback (most recent call last):
  File "chameleon_cli_main.py", line 8, in <module>
    import chameleon_cmd
  File "/home/jps/Downloads/ChameleonUltra/software/script/chameleon_cmd.py", line 6, in <module>
    from chameleon_utils import UnexpectedResponseError, expect_response
  File "/home/jps/Downloads/ChameleonUltra/software/script/chameleon_utils.py", line 45, in <module>
    def expect_response(accepted_responses: Union[int, list[int]]):
TypeError: 'type' object is not subscriptable

By looking around the history of chameleon_utils.py, I found a change at commit f6e2232 just near the reported error.

I checkout to the previous commit (a349288), build it, and 
python3 chameleon_cli_main.py
[Offline] chameleon --> hw connect
 { Chameleon connected } 
[USB] chameleon -->

Any help welcome...

JP

doegox, 2023-08-24:
bonjour JP :)
which Python version do you have ?
See https://github.com/RfidResearchGroup/ChameleonUltra/pull/66

> In its current state, the PR does not work on versions of Python prior to 3.9 because of type hinting used, if this is a problem and updating Python is not desired, those can be removed to allow it to work on at least 3.8 (maybe 3.7, not tested).

szikora, 2023-08-24:
Hi Doegox!

Nice to meet you here too... Yes, I know that my version of Debian is 2 versions behind... Python version is 3.7.3. Reading #66, it's clear. I know what to do now!

Regards,

closed 2023-08-24 by szikora