RfidResearchGroup/ChameleonUltra #91: Need Help About M1 Card Simulation

2212007054, 2023-08-24:
Hi,I just got my ChameleonUltra recently and using it to simulate my access card. After load the bin file dumped with pm3 easy, it works just fine when i use pm3 to validate the data. 
However, it can not pass the Access Control, so i use the pm3 to record the communications. Now i'm wondering whether the problem happens because of the latency during the simulation. Here is the trace file i got. 
[2023-08-24.zip](https://github.com/RfidResearchGroup/ChameleonUltra/files/12430262/2023-08-24.zip)

GameTec-live, 2023-08-24:
mifare classic 1k?

2212007054, 2023-08-25:
yep, it‘s mifare classic 1k. Here is the trace file I record  from physical card today.
[physicalcard.zip](https://github.com/RfidResearchGroup/ChameleonUltra/files/12435268/physicalcard.zip)

mazza73github, 2023-08-28:
Hi, in your physicalcard.trace the Reader is totally missing. Btw if you want i can give you a DFU fw to test if it fix your issue too.

2212007054, 2023-09-03:
After update the latest firmware, my question got solved, now the OEM reader can recognize ChameleonUltra.

closed 2023-09-03 by 2212007054