RfidResearchGroup/ChameleonUltra #92: Enhanced function of copying UID offline by pressing the A/B button

fd099, 2023-08-25:
First of all, thank you very much for your contribution. Now you can press the A/B button to copy UID offline. But I don’t know if it is possible to enter the 10/20/30 second waiting time after pressing the A/B button. During this waiting time, any card that enters the range can be copied. I think this feature is helpful in many scenarios. Thanks!

doegox, 2023-08-25:
can you explain such a scenario in a legit usage?

fd099, 2023-08-25:
> can you explain such a scenario in a legit usage?

For example, the non-standard size M1 card recognition distance is limited. I tried to use a strange angle to press the A/B button at a very close distance to recognize the card, and the A/B button is very small and hard. If the side of the LED light is attached to the card surface, it will be much better to press the A/B button, but the status of the indicator light cannot be seen.

doegox, 2023-08-25:
Ok then sth like 3s would be largely enough, which is the delay when in emulation mode before entering sleep mode.

fd099, 2023-08-25:
> Ok then sth like 3s would be largely enough, which is the delay when in emulation mode before entering sleep mode.

Thanks, that's a good idea too