RfidResearchGroup/ChameleonUltra #102: cmake is giving syntax error

moeyahia, 2023-08-26:
Hi Everyone,
I am trying to install this library, but i keep failing at the cmake command and i can't seem to understand what is going on. 

```
cmake . -D CMAKE_C_COMPILER="C:\Users\moeya\Desktop\Chamelion\ProxSpace\msys2\mingw64\bin\gcc.exe" -D CMAKE_CXX_COMPILER="C:\Users\moeya\Desktop\Chamelion\ProxSpace\msys2\mingw64\bin\g++.exe"
-- The C compiler identification is GNU 10.3.0
CMake Error at CMakeFiles/3.21.1/CMakeCCompiler.cmake:1 (set):
  Syntax error in cmake code at

    C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/3.21.1/CMakeCCompiler.cmake:1

  when parsing string

    C:\Users\moeya\Desktop\Chamelion\ProxSpace\msys2\mingw64\bin\gcc.exe

  Invalid character escape '\U'.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)
```

Below are the contents of CMakeLists.txt
```
cmake_minimum_required (VERSION 3.1)

project (mifare C)

set(EXECUTABLE_OUTPUT_PATH ${CMAKE_CURRENT_SOURCE_DIR}/../bin)
set(SRC_DIR ./)

set(COMMON_FILES
    ${SRC_DIR}/crapto1.c
    ${SRC_DIR}/crypto1.c
    ${SRC_DIR}/bucketsort.c
    ${SRC_DIR}/mfkey.c
    ${SRC_DIR}/parity.c)

include_directories(
    ${SRC_DIR}/
    )

# tools
add_executable(nested ${COMMON_FILES} nested.c)
add_executable(darkside ${COMMON_FILES} darkside.c)
add_executable(mfkey32 ${COMMON_FILES} mfkey32.c)
add_executable(mfkey32v2 ${COMMON_FILES} mfkey32v2.c)
add_executable(mfkey64 ${COMMON_FILES} mfkey64.c)

```

Where am i going wrong exactly?

Foxushka, 2023-08-26:
Run `cmake . -D CMAKE_C_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\gcc.exe" -D CMAKE_CXX_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\g++.exe"`, should work

moeyahia, 2023-08-26:
> cmake . -D CMAKE_C_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\gcc.exe" -D CMAKE_CXX_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\g++.exe"

It didnot

```
cmake . -D CMAKE_C_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\gcc.exe" -D CMAKE_CXX_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\g++.exe"
-- The C compiler identification is GNU 10.3.0
CMake Error at CMakeFiles/3.21.1/CMakeCCompiler.cmake:1 (set):
  Syntax error in cmake code at

    C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/3.21.1/CMakeCCompiler.cmake:1

  when parsing string

    C:\Users\moeya\Desktop\Chamelion\ProxSpace\msys2\mingw64\bin\gcc.exe

  Invalid character escape '\U'.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeOutput.log".
See also "C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeError.log".
```

GameTec-live, 2023-08-27:
> It didnot
>
> ```
> cmake . -D CMAKE_C_COMPILER="C:\\Users\\moeya\\Desktop\\Chamelion\\ProxSpace\\msys2\\mingw64\\bin\\gcc.exe" -D [...]

if your just compiling the cli binaries, why arent you just running cmake .

moeyahia, 2023-08-27:
> if your just compiling the cli binaries, why arent you just running cmake .

because it is giving me the same error
```

pm3 ~/ChameleonUltra/software/src$ cmake .
-- The C compiler identification is GNU 10.3.0
CMake Error at CMakeFiles/3.21.1/CMakeCCompiler.cmake:1 (set):
  Syntax error in cmake code at

    C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/3.21.1/CMakeCCompiler.cmake:1

  when parsing string

    C:\Users\moeya\Desktop\Chamelion\ProxSpace\msys2\mingw64\bin\gcc.exe

  Invalid character escape '\U'.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeOutput.log".
See also "C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeError.log".
```

Foxushka, 2023-08-27:
`git clean -d -f -x` maybe will fix it (warning: will remove all files that aren't present in repo!)

moeyahia, 2023-08-27:
> `git clean -d -f -x` maybe will fix it (warning: will remove all files that aren't present in repo!)

I already did a fresh install and it is still the same :(

GameTec-live, 2023-08-27:
> > `git clean -d -f -x` maybe will fix it (warning: will remove all files that aren't present in repo!)
> 
> I already did a fresh install and it is still the same :( . Trust me opening a ticket is the last resort i had. Even chatgpt didnt figure it out lol

try `pacman -Syu --noconfirm`, other than that i got no ideas...

moeyahia, 2023-08-27:
> pacman -Syu --noconfirm

So i did a git clean then ran the pacman update command and then navigated to src and executed cmake .

```
pm3 ~/ChameleonUltra/software/src$ cmake .
-- Building for: Ninja
-- The C compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: C:/Users/moeya/Desktop/Chamelion/ProxSpace/msys2/mingw64/bin/cc.exe
-- Check for working C compiler: C:/Users/moeya/Desktop/Chamelion/ProxSpace/msys2/mingw64/bin/cc.exe - broken
CMake Error at C:/Users/moeya/Desktop/Chamelion/ProxSpace/msys2/mingw64/share/cmake-3.21/Modules/CMakeTestCCompiler.cmake:69 (message):
  The C compiler

    "C:/Users/moeya/Desktop/Chamelion/ProxSpace/msys2/mingw64/bin/cc.exe"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeTmp

    Run Build Command(s):C:/Users/moeya/Desktop/Chamelion/ProxSpace/msys2/usr/bin/ninja.exe cmTC_59523 && [1/2] Building C object CMakeFiles/cmTC_59523.dir/testCCompiler.c.obj
    FAILED: CMakeFiles/cmTC_59523.dir/testCCompiler.c.obj
    C:\Users\moeya\Desktop\Chamelion\ProxSpace\msys2\mingw64\bin\cc.exe    -o CMakeFiles/cmTC_59523.dir/testCCompiler.c.obj -c 'C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeTmp/testCCompiler.c'
    /bin/sh: line 1: C:UsersmoeyaDesktopChamelionProxSpacemsys2mingw64bincc.exe: command not found
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeOutput.log".
See also "C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src/CMakeFiles/CMakeError.log".
```

This is the first error i receive, afterwhich i receive the parsing error. So it seems that this is the main issue, any thoughts?

doegox, 2023-08-27:
See here https://stackoverflow.com/questions/67273235/using-ninja-as-a-generator-with-msys2-mingw-on-windows-10-x64

Does it work if you change the ninja version?
```
pacman -R ninja
pacman -S mingw-w64-x86_64-ninja
```

moeyahia, 2023-08-27:
> See here https://stackoverflow.com/questions/67273235/using-ninja-as-a-generator-with-msys2-mingw-on-windows-10-x64
> 
> Does it work if you change the ninja version?
> 
> ```
> pacman -R ninja
> pacman -S mingw-w64-x86_64-ninja
> ```

You are a savior Sir 

```
pm3 ~/ChameleonUltra/software/src$ cmake .
-- Building for: Ninja
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 13.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/moeya/Desktop/Chamelion/ProxSpace/msys2/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done (3.9s)
-- Generating done (0.0s)
-- Build files have been written to: C:/Users/moeya/Desktop/Chamelion/ProxSpace/pm3/ChameleonUltra/software/src
```

closed 2023-08-27 by moeyahia

doegox, 2023-08-27:
@GameTec-live please check if guide should be adapted. I don't know if the problem occurs for everyone using proxspace or not