RfidResearchGroup/ChameleonUltra #105: mfkey32 crack don't find keys from reader (ultra and lite)

the0ne944, 2023-08-28:
Hello everyone, I have copied the UID of my protected Mifare 1k card using Chameleon Rev G by emulating DID and extracting the keys with MFKey32, it finds all the keys within a few seconds. However, with Chameleon Ultra and Lite, it only collects a few nonces (about 56) and then only finds FF keys, so it doesn't find the keys for my card. Is it possible to fix the problem? Thank you.

xianglin1998, 2023-08-28:
If your card secret key can be made public, I suggest that you download the detection data of your REVG through XMODEM, and then upload it to github for us to analyze and compare the differences.

the0ne944, 2023-08-28:
> If your card secret key can be made public, I suggest that you download the detection data of your REVG through XMODEM, and then upload it to github for us to analyze and compare the differences.

Hi, maybe I'm on the right way with the delay with NFC_MF1_FAST_SIM  active. This is the very first result, so it needs fine tuning because the pm3 trace is still not ever perfect. If you remember my previous video, before it was not responding at all; now, even it is still not perfetct, but  communicates correctly.

https://github.com/RfidResearchGroup/ChameleonUltra/assets/143393337/fc9773bc-995b-41cd-b5b8-4d58bade805a

[image]
[image]

As I supposed, now also the key extraction is quite fine, bacause it was related with that bad communication. Now it collect more nonces and find the correct key.

nieldk, 2023-08-29:
> > If your card secret key can be made public, I suggest that you download the detection data of your REVG through XMODEM, and then upload it to github for us to analyze and compare the differences.
> 
> Hi, maybe I'm on the right way with the delay with NFC_MF1_FAST_SIM  active. This is the very first result, so it needs fine tuning because the pm3 trace is still not ever perfect. If you remember my previous video, before it was not responding at all; now, even it is still not perfetct, but  communicates correctly.
> 
> 
> https://github.com/RfidResearchGroup/ChameleonUltra/assets/143393337/fc9773bc-995b-41cd-b5b8-4d58bade805a
> 
> [image]
> [image]
> 
> As I supposed, now also the key extraction is quite fine, bacause it was related with that bad communication. Now it collect more nonces and find the correct key.
> 
> Please add this fix to the firmware

You "could" create a PR with your changes?

doegox, 2023-08-29:
if creating a PR is too difficult, could you describe your changes? produce a diff?

xianglin1998, 2023-08-29:
Yes, I need to further confirm where the problem lies, as theoretically all communication is within the timing window.

xianglin1998, 2023-08-29:
> Yes, I need to further confirm where the problem lies, as theoretically all communication is within the timing window.

Is it because the ChameleonUltra is not a standard card that causes this issue? Or because the reader is not a standard MifareClassic reader and the processing speed is slow that causes this problem? This all requires investigation.

doegox, 2023-08-29:
BTW, sharing a nice project @gentilkiwi reminded me about : https://github.com/josevcm/nfc-laboratory which allows to sniff very easily a HF communication with a SDR and decode it. I've still to find time to dig into it and find a reader that gets trouble with Ultra emulation...

the0ne944, 2023-08-29:
> BTW, sharing a nice project @gentilkiwi reminded me about : https://github.com/josevcm/nfc-laboratory which allows to sniff very easily a HF communication with a SDR and decode it. I've still to find time to dig into it and find a reader that gets trouble with Ultra emulation...

Hi, we preparing me and Paul what you need, soon he post the issue and I link it here. Thanks

xianglin1998, 2023-08-29:
> 

Good!!!

the0ne944, 2023-08-29:
> > BTW, sharing a nice project @gentilkiwi reminded me about : https://github.com/josevcm/nfc-laboratory which allows to sniff very easily a HF communication with a SDR and decode it. I've still to find time to dig into it and find a reader that gets trouble with Ultra emulation...
> 
> Hi, @xianglin1998   we preparing me and Paul what you need, soon he post the issue and I link it here. Thanks

the0ne944, 2023-08-29:
@doegox @xianglin1998 here the full explanation

https://github.com/RfidResearchGroup/ChameleonUltra/issues/110

the0ne944, 2023-08-29:
> @the0ne944 thank you! so is it ok to close this one and concentrate the solving on the other one?

Yes sir, I hope you fix this big problem! Thank you soon much

closed 2023-08-29 by doegox