RfidResearchGroup/ChameleonUltra #106: Getting worst after update latest firmware

vinpme, 2023-08-28:
![Uploading 263702988...]
Before update it was smooth, no more delay. But update firmware it really hard to detect the reader.

shallax, 2023-08-28:
User reports that since upgrading to the latest firmware via the GUI from @GameTec-live, reader detection is poor. See the attached videos of a Chameleon Ultra vs Flipper Zero 

https://github.com/RfidResearchGroup/ChameleonUltra/assets/15132969/c3f6d328-77a7-4d64-a9cd-cf2d07f23978
https://github.com/RfidResearchGroup/ChameleonUltra/assets/15132969/bee00351-ae65-4009-89df-0bb6b5cc563e

Apparently this was not the case with stock firmware.

GameTec-live, 2023-08-28:
> User reports that since upgrading to the latest firmware via the GUI from @GameTec-live, reader detection is poor. See the attached videos of a Chameleon Ultra vs Flipper Zero [...]

i doubt that its the guis fault because it just takes the fw from gh

shallax, 2023-08-28:
User has tried disabling animations to see if this makes any difference: 

https://github.com/RfidResearchGroup/ChameleonUltra/assets/15132969/679c74fd-7ce6-422b-afe5-5e44048fdc24

shallax, 2023-08-28:
Reader: https://shop.linkserve.mv/products/soca-em-card-reader

shallax, 2023-08-28:
It was suspected it might be a clash between HF and LF in a single slot, so here is Slot 8 with only LF: 

https://github.com/RfidResearchGroup/ChameleonUltra/assets/15132969/5beaf497-7b0b-48b1-9e29-05a8eae16e14

xianglin1998, 2023-08-29:
The blue light indicates that the LF simulation is working. From the factory firmware to the latest firmware, the LF simulation has been modified many times, which may have caused compatibility loss of some readers.

spp2000, 2023-08-30:
i don't know if it can be useful, but if you look at the white RF LED, it doesn't always turn on and sometimes it turns off immediately instead of staying on

vinpme, 2023-09-14:
> The blue light indicates that the LF simulation is working. From the factory firmware to the latest firmware, the LF simulation has been modified many times, which may have caused compatibility loss of some readers.

how can I fix this ya?