RfidResearchGroup/ChameleonUltra #111: Mifare 1k emulation problem

mitmarcus, 2023-08-29:
Keyfob Im trying to copy is a "NXP MIFARE"

The reader is a Salto Modular XS - EU Wall Reader

Card is successfully read and the keys are aqquired, however upon emulation the reader always denies the chameleon. Running mfkey32 results in keys that are always "FFFFFFFFFFFF".

I have the binary dump, but can't upload it here. Am I missing something to emulate it?

the0ne944, 2023-08-29:
> Keyfob Im trying to copy is a "NXP MIFARE"
>
> The reader is a Salto Modular XS - EU Wall Reader
>
> Card is successfully read and the keys are aqquired, however upon emulation the reader always denies [...]

No, there is a firmware problem your re Right 

> Keyfob Im trying to copy is a "NXP MIFARE"
>
> The reader is a Salto Modular XS - EU Wall Reader
>
> Card is successfully read and the keys are aqquired, however upon emulation the reader always denies [...]

Hi we have all the same problem refer: https://github.com/RfidResearchGroup/ChameleonUltra/issues/110

doegox, 2023-08-29:
This is very probably the same as #110 so I will close this one as duplicate and invite you to discuss it on the other issue, thank you!

closed 2023-08-29 by doegox