RfidResearchGroup/ChameleonUltra #113: compiling with VS 2022 Community - error on darkside

plato79, 2023-08-29:
Well, with latest commit [f80e7fe](https://github.com/RfidResearchGroup/ChameleonUltra/commit/f80e7fe14e2eb6827499dbd3ebf8fd1d92d8abb9) while I could compile the mfkey and nested, I get an error when compiling darkside
```
Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	1 unresolved externals	darkside	C:\Users\Hakan\source\repos\ChameleonUltra\software\bin\Release\darkside.exe	1	
Warning	C6011	Dereferencing NULL pointer 'last_keylist'. 	darkside	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\darkside.c	107	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32.c	27	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32.c	28	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32.c	29	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32.c	30	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32.c	31	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32.c	32	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	29	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	30	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	31	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	32	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	33	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	34	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey32v2	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey32v2.c	35	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	44	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	45	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	46	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	47	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	48	
Warning	C6031	Return value ignored: 'sscanf'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	53	
Warning	C26451	Arithmetic overflow: Using operator '*' on a 4 byte value and then casting the result to a 8 byte value. Cast the value to the wider type before calling operator '*' to avoid overflow (io.2).	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	53	
Warning	C6385	Reading invalid data from 'enclen'.	mfkey64	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\mfkey64.c	67	
Warning	C6001	Using uninitialized memory 'bucket.head'.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crapto1.c	192	
Warning	C6262	Function uses '262288' bytes of stack.  Consider moving some data to heap.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crapto1.c	225	
Warning	C6001	Using uninitialized memory 'table'.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crapto1.c	250	
Warning	C6292	Ill-defined for-loop.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crapto1.c	385	
Warning	C26451	Arithmetic overflow: Using operator '<<' on a 4 byte value and then casting the result to a 8 byte value. Cast the value to the wider type before calling operator '<<' to avoid overflow (io.2).	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crypto1.c	46	
Warning	C6297	Arithmetic overflow.  Results might not be an expected value.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crypto1.c	46	
Warning	C26451	Arithmetic overflow: Using operator '<<' on a 4 byte value and then casting the result to a 8 byte value. Cast the value to the wider type before calling operator '<<' to avoid overflow (io.2).	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crypto1.c	47	
Warning	C6297	Arithmetic overflow.  Results might not be an expected value.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\crypto1.c	47	
Warning	C6001	Using uninitialized memory 'keys'.	nested	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\nested.c	201	
Error	LNK2019	unresolved external symbol compare_uint64 referenced in function main	darkside	C:\Users\Hakan\source\repos\ChameleonUltra\software\src\build\darkside.obj	1	
```

What am I doing wrong? The other executables are all fine.

doegox, 2023-08-29:
Please provide a full log including the output of `cmake` and of `make`, so we can see at least which compiler you are using.

GameTec-live, 2023-08-29:
windows 11 pro, cmake 3.26.4 and Visual Studio 17 2022 works fine for me

plato79, 2023-08-30:
hmm. yes, it seems cmake itself can build but if you open the solution it cannot. weird.