RfidResearchGroup/ChameleonUltra #127: BLE connection not possible

foXaCe, 2023-09-05:
impossible for me to connect with my android 13 phone
no problem with my android 13 tablet
both are Oppo brand
I tried everything 🤷‍♂️

mitmarcus, 2023-09-05:
try turning on location

foXaCe, 2023-09-05:
I enabled location

foXaCe, 2023-09-05:
here is in order what is happening

[image]
[image]
[image]

foXaCe, 2023-09-05:
at the end he disappears

Markg546, 2023-09-05:
unpair from device settings and/or clear bonds from the chameleon.

foXaCe, 2023-09-05:
I did that too.  several times

mitmarcus, 2023-09-05:
and does it ask you to input the pass 123456?

foXaCe, 2023-09-06:
yes

foXaCe, 2023-09-06:
yes the LEDs go out. I already have the latest firmware 47aed37

Foxushka, 2023-09-06:
Even if you smash buttons all this time? It goes down at moment where connection popup appears?

xianglin1998, 2023-09-08:
@foXaCe The latest firmware has been disable pair by default. How to update your device firmware: https://rfidresearchgroup.github.io/ChameleonUltra/firmware.html

foXaCe, 2023-09-08:
hello after upgrade it work
thanks you