RfidResearchGroup/ChameleonUltra #128: Add Github Pages deployment for docs

GameTec-live, 2023-09-06:
As this repository already owns the chameleonultra.com domain it might be useful to enable a github pages deployment and point it to the docs folder. 

[image]

This would automatically set up a Markdown web builder. One could then link this markdown web builder to a subdomain (like docs.chameleonultra.com) by creating a CNAME file and have a quick and short link to web docs.

Benefits:
No practical ones, but its pretty, a short link and its "cleaner" without all of the github interface. It might also load faster, but that doesnt really matter, does it? + Its literally free...

xianglin1998, 2023-09-07:
Good idea, try it.

xianglin1998, 2023-09-07:
Finish, document link to here:
https://chameleonultra.com/docs

closed 2023-09-07 by xianglin1998