RfidResearchGroup/ChameleonUltra #135: [Theoretical Idea] CU auto detects reader

mitmarcus, 2023-09-08:
_If it's not possible just close it. If it doesnt make sense because the reader will first deny the connection and only then it works (aka takes more time than just pushing a button) also close it._

In theory could the CU detect the reader and change to the appropriate card in the slot? 
Lets say you save the card and you approach it to the reader to save the reader as well with it. 

So whe CU is near the reader it switches to the correct card based on the first few keys from reader.

doegox, 2023-09-08:
In most readers, key depends on the UID and as you don't know yet which UID to present to the reader during the anticollision phase, you can't recognize the reader, even after a first failed auth.

mitmarcus, 2023-09-08:
Thanks for explaining

closed 2023-09-08 by mitmarcus