RfidResearchGroup/ChameleonUltra #137: Add a long press action to write to the card from the current slot [idea]

bvasya, 2023-09-08:
**Is your feature request related to a problem? Please describe.**
It is often necessary to update the ID on the card offline

**Describe the solution you'd like**
Add a long press action to write to the card from the current slot

**Describe alternatives you've considered**
Store the chip type (LF t55xx/em4x05, HF gen1/gen2) in the action or slot settings

Markg546, 2023-09-08:
Isn't there such a feature already?

afaik you need a phone to "crack" the key pair

cyber-vi-king, 2023-09-24:
there is long press to save uid to current slot. I'm not sure if I'm understanding this correctly, but what could be useful is another setting to write to a t5577 from the current slot. 

As in:
- long press A while holding card against ultra -> uid from card is now saved to / emulated on selected slot
- long press B while holding ultra against empty t5577 -> uid from selected slot gets written to empty t5577
currently that would involve the ultra connected to MTools app or CLI (for the write part)

psxdev1, 2024-06-05:
This would be a great addition