RfidResearchGroup/ChameleonUltra #138: Unable to clear ble bonds after update to v1.0

georgikoemdzhiev, 2023-09-08:
***Things to try before submitting a bug report***

* read the [troubleshooting guide](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/troubleshooting.md)
* check existing [issues](https://github.com/RfidResearchGroup/ChameleonUltra/issues)
* use the latest firmware and CLI
* for issues specific to another client than the Python CLI, use the corresponding issue tracker. E.g. [here](https://github.com/GameTec-live/ChameleonUltraGUI/issues) for the ChameleonUltraGUI

***Compilation problems***
Try compiling with verbose. Use `make VERBOSE=1` for the firmware and the software/src tools.
Include the verbose compilation logs.

***flashing problems***
Have you followed the instructions properly?

**Describe the bug**
Unable to clear ble bonds after update to v1.0. I am also unable to connect to the device using the ChamelionUltraGUI app as well.

**To Reproduce**
1. I updated my ChamelionUltra to the latest version using the ChamelionUltraGUI app
2. I connected the device to my PC and connected it using Serial
3. I executed `hw ble bonds clear` 

**Expected behavior**
The Bluetooth bonds get cleared without errors

**Actual behaviour**
I am getting this error:
```sh
[USB] chameleon --> hw ble bonds clear
Serial Error device reports readiness to read but returned no data (device disconnected or multiple access on port?), thread for receiver exit.
```

**Screenshots**
n/a

**Host (please complete the following information):**
 - OS and version - endeavours, 2023.03.26
 - for compilation issues, the toolchain version
 - inside CLI run `hw version` and paste the output here - Version: v1.0 (0dc83b4)
 
**Additional context**
After I "Forgot" the device on my phone and retried connecting to the chamelionUltra it was able to connect to it without issues.

xianglin1998, 2023-09-08:
This is a bug, and pairing is disabled by default in the new firmware. In this case, the API that clears binding records should unconditionally return success without performing any operations, otherwise it will cause the chameleon to crash.

aramova, 2023-09-08:
On a Pixel 7 device, all BT pairings removed with the latest GUI (Commit 290) fresh install with no data or cache in place and Ultra firmware Version: v1.0 (dev-142-g0dc83b4-dirty the following occurs: 

Connected via USB with cli: 
1. Enable ble pairing via CLI
2. Scan with GUI on Phone
3. Attempt to pair via the GUI
4. CLI shows the subsequent "No task wait process $NNNN" values shown below
5. GUI fails as timeout waiting for response to command 1000

`[USB] chameleon --> hw settings blepair
 - Is ble pairing enable: No
[USB] chameleon --> hw settings blepair -e 1
 - Is ble pairing enable: No
 - Successfully change ble pairing to Enable state.
[USB] chameleon --> hw settings store
Storing settings...
 - Store success @.@~
[USB] chameleon --> No task wait process: $1019
No task wait process: $1025
No task wait process: $1000
[USB] chameleon --> hw version
 - Version: v1.0 (dev-142-g0dc83b4-dirty)`

doegox, 2023-09-08:
I think you need to quit the CLI before attempting to connect via the GUI

aramova, 2023-09-08:
You're correct. It's odd cause it worked in prior versions, I went back and checked some screen recordings I was making for firmware interaction how-tos. Maybe it was a bug prior allowed for that behavior? 

I'll withdraw my report as this appears to be WAI.

xianglin1998, 2023-09-14:
**Fixed by** [98605be](https://github.com/RfidResearchGroup/ChameleonUltra/commit/98605be5b8cf93bf6e54e8f84dab960dd8879f9a)

In this commit, must initialize the pairing manager. However, whether to handle BLE events in the pairing manager is determined by the setting parameters loaded during the first wake-up.

closed 2023-09-14 by xianglin1998