RfidResearchGroup/ChameleonUltra #139: [Help me] My chameleon is bricky

qingyifeihu, 2023-09-08:
1. I used the Chameleon Ultra GUI to do a firmware update for the chameleon.

2.The chameleon hardware started flashing two green lights, I waited for about 20 minutes, the green light was still flashing, I unplugged the chameleon hardware.

3.The chameleon hardware started flashing the two red lights in the middle, I plugged the chameleon hardware back in, and the software interface showed "Tihs probably means your firmware is corrupted." Do you want to flash latest FW?”

4.Click on the "Flash" software below and say "Downloading and preparing new Chameleon Ultra firmware..." The chameleon hardware and software didn't respond and I got stuck in the loop.

Does anyone know how to re-flash the firmware into the chameleon? Thank you!!!

m-kozlowski, 2023-09-08:
use nrfutil and flash over usb

hank9999, 2023-09-08:
according to my test
red lights flashing -> in dfu mode, waiting to connect
green lights flashing -> connected, but not writing
blue lights flashing -> writing firmware

you may can try disconnecting the usb and waiting for timeout to check whether chameleon can start.
if it still does not start, you need to follow the process in [docs/firmware](https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/firmware.md) to flash your chameleon with nrfutil

qingyifeihu, 2023-09-09:
> 

Ok, I'll give it a try, thanks

qingyifeihu, 2023-09-09:
Ok, I'll give it a try, thanks

Foxushka, 2023-09-10:
If you tried to flash on iOS, there was many DFU bugs that got fixed in last releases. If other platform - report to our repo, logs in Debug menu also will be useful

Foxushka, 2023-09-11:
BLE is fine now, it has CRC and other checks

doegox, 2023-09-27:
what is the situation? can we close this ?