RfidResearchGroup/ChameleonUltra #144: Mfc 1k full emulation issue with an OEM terminal

spp2000, 2023-09-11:
An OEM access control system, in addition to simply allowing access, also allows you to perform a series of operations from the terminal (queries, etc.). To do this, the tag must be left on the reader in order to keep the session open.

Analyzing 3 traces I noticed that, after the first authentication, the system waits for instructions by executing a loop of two `READBLOCK`: it continuously asks for `AUTH-B(12)` - `READBLOCK(12)` - `READBLOCK(13)` (see pictures).

With Ultra (latest fw) the initial authentication works fine but, during the wait/loops, every 2 seconds (more or less) the communication stops and starts again from the beginning.

For the 3 traces acquired we observe that:
- after 3 loops, _fdt_ after `AUTH-B(12)` is shorter than the previous (about 1600 Vs 3000) -> collision?
- after 225 or 230 loops (about 2 seconds), _fdt_ after `READBLOCK (13)` are shorter than the previous (about 1600 Vs 5200) -> collision?
Then we have a new ANTICOLL, etc... 

I can't acquire more time with pm3, but from the terminal I see that this behavior repeats continuously after about 2 seconds, therefore prevents me from continuing with the next operations.

Given that this behavior does not seem exactly systematic, I will try to take other traces to have a minimum of statistics on this behavior, if it can be useful.
Thank you
traceman ;)

**Trace n.1**

[image]

**Trace n.2**

[image]

**Trace n.3**

[image]

mitmarcus, 2023-09-14:
can confirm there's still an issue 

https://github.com/RfidResearchGroup/ChameleonUltra/assets/114725463/ff0c4722-ba84-4b41-85f7-ac942d792f5f

spp2000, 2023-09-14:
I took more than 10 traces and I can confirm that, in my case, the communication restarts **always and only** after those commands during the waiting loops. fdt is **always** about 1600 when the issue occurs, but there is no rule as to when this happens: it can happen after 2, 30, 50, 70, 200 loops indifferently.

spp2000, 2023-09-14:
**Ultra Vs Rev.G**
Rev.G fdt never drops (like Ultra in the pictures above) and are much longer

[image]

[image]

mitmarcus, 2023-09-28:
With 2.0 it seems to work
@spp2000 can you check on your readers?

mitmarcus, 2023-10-11:
aaaand it's broken again

spp2000, 2023-10-11:
For me nothing has changed since the last time the emulation was improved.  The readers that worked continue to work, while the one that was ko remained ko.
What problem do you experience?

mitmarcus, 2023-10-11:
remember my "problematic" Salto modular XS reader?
with the original V2 (2 weeks ago) it was working without issue.
I have updated today to the latest and reader won't work again with it.

really no idea what the issue might be.

xianglin1998, 2023-10-19:
@mitmarcus After tracing with SPP2000, we have finally resolved two bugs that affect the simulation of 14a and mf1. You can update to the latest build version to test whether they have solved your problem at the same time. I think this is a high probability.

closed 2023-10-19 by xianglin1998