RfidResearchGroup/ChameleonUltra #146: Chameleon Lite / DevKit - LF emulation not working

panayotoff, 2023-09-11:
Hi all, 
I've received my Chamaleon Lite and Devkit from the Indiegogo campaign. I have troubles with emulating the LF on Lite and Devboard.

**Describe the bug**
### I've tried the following:

- Flash latest FW ( both from the GUI and CLI )
- Set only LF on slot, setting LF and HF on the same slot
- Waking up chameleons beforehand 
- Trying different distances. 

### Readers
- nothing on Flipper0
- nothing on my home card-access reader
- nothing on PM3 when I try general `lf search` . If however I'm more specific and I try `lf em 410x reader -@` on PM3 I would get the proper ID

**To Reproduce**
Steps to reproduce the behavior.  

1. Install latest FW ( and any previous for this matter )
2. Read LF EM410x card with Ultra ( reader mode is working )
3. Copy the EM410x card to a slot ( either GUI or CLI )
4. Go to this slot 
5. present to reader

**Expected behavior**
Reader is able to read the card ( Tiny and DevKit )

**Screenshots**

**Additional context**

xianglin1998, 2023-09-12:
1. Have you attempted to execute the hw factory_reset command after update to the latest firmware.(Updating firmware may damage simulated data)
2. Is the type of simulation card correctly set in the card slot? using cmd: `hw slot type`
3. Did you see the blue light of the chameleon when reading the card? If not, it may be that your card slot selection is incorrect. Please select the correct card slot for your uploaded ID.

xianglin1998, 2023-09-12:
From the factory firmware to the latest firmware, the low-frequency simulation has been modified many times. If your chameleon cannot correctly read the card number even when the blue light is on during card reading, please try using the factory firmware test.

panayotoff, 2023-09-12:
Thanks, I did a upgrade to latest (4d8fb87) and did a factory reset, which has not fixed the issue. After factory resetting, I can not read with my PM3 or F0 slots 1 or 3 in LF ( HF is readable thought ). The DevKit is able to read LF cards, but not emulate them back. 

I've set the slot type to LF ( `hw slot type -t 1 -s 1` for first slot, only EM410X there, saved to flash ) and still does not work. 

When I read the card I don't see blue light from the non-slot LEDs
On the Tiny, I can see the light next to the USB-C port ( first side LED ). }
On the DevKit, the central LED ( between the version number and chameleon dev edition print ) lights up. 
But both light in white, not blue.

xnfra, 2023-09-16:
I am having a similar problem simply getting the chameleon to read lf t5577 cards. Does low frequency even work? Is my device defective?