RfidResearchGroup/ChameleonUltra #148: [idea] Possible enhancement: energy saving mode (aka stealth mode)

spp2000, 2023-09-12:
Hi all!
I was thinking of a mode that allows the user to turn off all the slot LEDs, exclusively when Chameleon is connected (to the App/CLI).

Why?
This would allow to reduce energy consumption a little (especially when connected with BLE, without USB) but also (or mainly) to have a sort of stealth mode in order to be more discreet.
For the battery saving reason, I think of all the time when you use it as a reader or to find keys, etc while in BLE.
The only information you would lose in this mode is the active slot when looking at the device, but you can easily find it out from the App or CLI. Furthermore, during these operations it's not important to know the active slot.

How?
It could be a button in the App and a command in the CLI, which turns off all the LEDs.
This mode should NOT be saved as a setting in the flash, so the next time you turn on Chameleon it returns to "normal" mode. By doing so, there is no risk of having a dark Chameleon that could cause confusion for the less experienced.

I don't know if it makes sense for you and, above all, if it's difficult to implement due to how the current firmware is written.

I ask you not to trash this idea for the sole reason that there is a lot of work to do at the moment. If that's the only reason, let's save it for the future. Personally I find it useful.
Thank you for your time

LupusE, 2023-09-18:
While browsing through the settings, I've seen the animation settings:
```
[USB] chameleon --> hw settings animation set -h
usage: hw settings animation set [-h] -m {0,1,2}

Please enter correct parameters

options:
  -h, --help            show this help message and exit
  -m {0,1,2}, --mode {0,1,2}
                        0 is full (default), 1 is minimal (only single pass
                        on button wakeup), 2 is none
```

It should be `hw settings animation set -m 2`. But to be honest, I don't know if this only reduce the blinking or shut down the disco mode.
I like your concerns, that a fully silent Chameleon can easily confused with an offline one or empty battery.

spp2000, 2023-09-22:
> It should be `hw settings animation set -m 2`. But to be honest, I don't know if this only reduce the blinking or shut down the disco mode. I like your concerns, that a fully silent Chameleon can easily confused with an offline one or empty battery.

Yes, this refers to the animations after wake-up and befere sleep, while I would activate this mode when Chameleon is on and connected to a smartphone/pc