RfidResearchGroup/ChameleonUltra #153: Python client shows incorrect MIFARE emulator configuration

0xDRRB, 2023-09-16:
Hi,

I activated the "anti-collision data from block 0" functionality for slot 6.  But :

```
[USB] chameleon --> hw slot change -s 1
 - Set slot 1 activated success.

[USB] chameleon --> hw slot list
 - Slot 1 data (active):
   HF: lav - Mifare Classic 4k
   LF: Empty - Unknown
 - Mifare Classic emulator settings:
   Detection (mfkey32) mode: disabled
   Gen1A magic mode: disabled
   Gen2 magic mode: disabled
   Use anti-collision data from block 0: disabled   <-----------------------------
   Write mode: Normal
 - Slot 2 data (disabled):
   HF: Empty - Unknown
   LF: Empty - EM410X
 - Slot 3 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 4 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 5 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 6 data:
   HF: mflav4k - Mifare Classic 1k
   LF: Empty - Unknown
 - Mifare Classic emulator settings:
   Detection (mfkey32) mode: disabled
   Gen1A magic mode: disabled
   Gen2 magic mode: disabled
   Use anti-collision data from block 0: disabled   <-----------------------------
   Write mode: Normal
 - Slot 7 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 8 data:
   HF: plopHF - Mifare Classic 1k
   LF: plopLF - Unknown
 - Mifare Classic emulator settings:
   Detection (mfkey32) mode: disabled
   Gen1A magic mode: disabled
   Gen2 magic mode: disabled
   Use anti-collision data from block 0: disabled   <-----------------------------
   Write mode: Normal
[USB] chameleon -->
```

The functionality seems disabled for slots 1, 6 and 8. But if I switch to slot 6:

```
[USB] chameleon --> hw slot change -s 6
 - Set slot 6 activated success.
[USB] chameleon --> hw slot list
 - Slot 1 data:
   HF: lav - Mifare Classic 4k
   LF: Empty - Unknown
 - Mifare Classic emulator settings:
   Detection (mfkey32) mode: disabled
   Gen1A magic mode: disabled
   Gen2 magic mode: disabled
   Use anti-collision data from block 0: enabled   <-----------------------------
   Write mode: Normal
 - Slot 2 data (disabled):
   HF: Empty - Unknown
   LF: Empty - EM410X
 - Slot 3 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 4 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 5 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 6 data (active):
   HF: mflav4k - Mifare Classic 1k
   LF: Empty - Unknown
 - Mifare Classic emulator settings:
   Detection (mfkey32) mode: disabled
   Gen1A magic mode: disabled
   Gen2 magic mode: disabled
   Use anti-collision data from block 0: enabled   <-----------------------------
   Write mode: Normal
 - Slot 7 data (disabled):
   HF: Empty - Unknown
   LF: Empty - Unknown
 - Slot 8 data:
   HF: plopHF - Mifare Classic 1k
   LF: plopLF - Unknown
 - Mifare Classic emulator settings:
   Detection (mfkey32) mode: disabled
   Gen1A magic mode: disabled
   Gen2 magic mode: disabled
   Use anti-collision data from block 0: enabled   <-----------------------------
   Write mode: Normal
```

Now the functionality seems activated for all 3 slots.

At fisrt I even thought that the configuration of the MIFARE emulator was global to all slots. It's not.

doegox, 2023-09-17:
Indeed it was a mess. Config is per slot but should be shown only for the active slot and only if it's a MF1 type.
I rewrote it in my big PR, it looks like this now:
[image]

xianglin1998, 2023-09-27:
@doegox Does this issue need to be closed?

doegox, 2023-09-27:
yes it's merged now

closed 2023-09-27 by doegox