RfidResearchGroup/ChameleonUltra #154: Enabling Shadow mode does not 'stick'

bpinard, 2023-09-20:
"Shadow" mode setting does not stick.

I have tried with an iOS 17 device using bluetooth, an android device using USB-C, and the same android device using bluetooth.

The app used in all cases was CU GUI.

Chameleon Ultra firmware is 1.0 (98605be)

Upon setting slot #1 to Shadow mode, the setting does not save properly.  The other modes seem to be able to be set properly.
Thank you.

GameTec-live, 2023-09-20:
seems to work fine on cli, so it looks like a GUI issue, please move the issue to the CU GUI issue tracker

Foxushka, 2023-09-23:
PoC to reproduce on CLI:

```[USB] chameleon --> hw raw -c 4017 -d 03
 - Received:
   Command: 4017
   Status:  0x68 STATUS_DEVICE_SUCCESS: Device operation succeeded
   Data (HEX): 
[USB] chameleon --> hw slot list
 - Slot 1:(active)
   HF:  Mifare Classic 1k
    - Mifare Classic emulator settings:
      Detection (mfkey32) mode:               enabled
      Gen1A magic mode:                       disabled
      Gen2 magic mode:                        disabled
      Use anti-collision data from block 0:   disabled
      Write mode:                             Shadow
...
[USB] chameleon --> hw slot change -s 2
 - Set slot 2 activated success.
[USB] chameleon --> hw slot change -s 1
 - Set slot 1 activated success.
[USB] chameleon --> hw slot list
 - Slot 1:(active)
   HF:  Mifare Classic 1k
    - Mifare Classic emulator settings:
      Detection (mfkey32) mode:               enabled
      Gen1A magic mode:                       disabled
      Gen2 magic mode:                        disabled
      Use anti-collision data from block 0:   disabled
      Write mode:                             Deceive
   LF:  EM410X
...
```

doegox, 2023-09-24:
Hmm it's not easy to fix, shadow = don't save changes so... shadow setting is not saved :facepalm: 
This will require some deeper changes in the internal logic to handle shadow properly

doegox, 2023-09-24:
ok I'll solve it by creating a SHADOW_REQ indicating a shadow mode is requested, but still needs to be stored.
[image]

doegox, 2023-09-24:
Fixed in https://github.com/RfidResearchGroup/ChameleonUltra/pull/147/commits/1dc8891e1b3eb122ff657f219393a8aa1ad0ae1c
(branch not yet merged in main, waiting for all clients to be ready for all big changes of that branch)

doegox, 2023-09-24:
Note that when in shadow, it does not store any change to the MF1 including its settings.
So e.g. if one wants to change gen1a he needs to do
```
hf mf settings --gen1a 1
hf mf settings --write 3     <-- this will move it to SHADOW_REQ and authorize the save to flash
hw slot update
```

doegox, 2023-09-27:
Now merged in the v2.0.0