RfidResearchGroup/ChameleonUltra #162: Improvement suggestions from old users

lin1414814113, 2023-10-02:
Previously, Tinypro was used, but now I hope to add the old features of Tinypro. The first one is to scan dump files in specified folders in batches and automatically synchronize file names as card names, so that there is no need to manually add them one by one. The second one was that before, Tinypro could automatically connect by opening the software with Bluetooth on the phone, without pressing the physical button of Tinypro. Now, UItra still needs to press the chameleon to connect to Bluetooth, and UItra swipes its card slower than Tinypro. Let's see if we can improve it. Just refer to the previous Tinypro to see if we can improve it

之前是用tinypro，，现在这个希望增加之前tinypro的旧功能，第一个：可以批量扫描指定文件夹里面的dump文件，并且自动同步文件名作为卡片名，这样就不用一个一个手动添加。第二个之前tinypro开着手机蓝牙打开软件可以自动连接，不用按tinypro的物理按钮，现在UItra还需要按一下变色龙才能链接蓝牙，，并且UItra刷卡也比tinypro慢，看看能不能改进，参照之前tinypro的就很好用看看能不能改进

GameTec-live, 2023-10-02:
the tiny and ultra are completly different hardware

lin1414814113, 2023-10-02:
@GameTec-live 
So can Ultra add the functionality of the previous Tiny mobile application, the first scanner function, to batch scan dump files. The second step is to open the mobile CU GUI application and automatically connect without pressing the ultra physical button

GameTec-live, 2023-10-02:
you have to press the physical buttons to wake the device and thats not gonna change any time soon, because thats the energy saving feature that makes the CU usable with such a smal battery...

lin1414814113, 2023-10-02:
@GameTec-live But the previous Tiny is also quite energy-efficient, or if we choose a setting, we need to press the physical button to connect ourselves. In addition, currently using the (EM410X) low-frequency card makes it difficult to open the firewall, and occasionally it can be opened. Can we update it together,