RfidResearchGroup/ChameleonUltra #163: [idea] Battery level with "slot" LEDs

spp2000, 2023-10-02:
It could be useful to check battey level also when Chameleon is in our pocket, not only when connected with tha app or cli.
A function to display battery level using the 8 "slot" LEDs that can be triggered, for example, with a long press button.
I would use a different color for LEDs (white like Chameleon Mini?)
Example: if battery is 25%, long press B should light 1 and 2 LEDs with white color, then Chameleon sleeps after few seconds.

doegox, 2023-10-02:
To avoid sacrificing a long press on that, it could be done systematically e.g. after the shutdown sequence, blink briefly the LEDs to indicate the battery level. (also apparently there is some limitation that battery level is provided only 4 secs after boot so we can't really wakeup and get the battery level immediately)
The white with these rgb doesn't look great imho, maybe cyan.

I had the impression that the 4 pulsing leds when connecting via USB without client were meant to indicate battery level as well with 25-50-75-100 but implementation not being finished. I don't remember, I think some source code comments were in that direction... So we could make it work and use the same led convention for both situation.

doegox, 2023-10-02:
ha, here (auto translated from Chinese) in rgb_marquee.c
```c
// Charging animation
// the current percentage of the battery 0-4 4 represents full electric breathing light
```

GameTec-live, 2023-10-03:
as currently both longpress buttons do the same thing, id say why not use one of them for battery info? they are userdefineable anyways...

spp2000, 2023-10-03:
> To avoid sacrificing a long press on that, it could be done systematically e.g. after the shutdown sequence, blink briefly the LEDs to indicate the battery level. (also apparently there is some [...]

ok, if we need 4-5 secs to have a correct measure it could be nice also just before sleeping

> The white with these rgb doesn't look great imho, maybe cyan.

I totally agree. I really like both the cyan and fuchsia that you see when the chameleon is charging (only)

> I had the impression that the 4 pulsing leds when connecting via USB without client were meant to indicate battery level as well with 25-50-75-100 but implementation not being finished. I don't [...]

ok..but it could be useful to have a better resolution of the charge level using all the 8 leds (for example when the battery starts to lose its capacity, with more LEDs you can understand it better).
BTW any battery level implementation is fine for me. ;-)

doegox, 2023-10-04:
@GameTec-live reminded me that with the current hardware design, we can't measure the battery when charging over USB.
So what remains possible to implement is battery level when offline.
With a long press (but info available only 4s after boot) or on sleep, both could be made available and configurable.
Maybe that 4s delay may be reduced, I didn't look into it.

doegox, 2023-10-04:
https://github.com/RfidResearchGroup/ChameleonUltra/commit/702dba0d9310a0d6686377f926eb3c262f7beb7a
you may pull and try

spp2000, 2023-10-04:
Great @doegox!! The 8 cyan LEDs are beautiful!
Also the flashing red LEDs (if I ask the battery level too early) are nice!
Everything seems to be working fine for me! Thanks

PS: now I have to let the battery run down a bit to see them turn off :D

closed 2023-10-04 by doegox