RfidResearchGroup/ChameleonUltra #164: Reworking CLI commands

doegox, 2023-10-06:
Historically, 99% of CLI commands reflect directly the firmware protocol, but it does not have to. It has to be user-friendly.

Note that with the current dev CLI, we can already provide `-h` to any command (as in the Proxmark3 client).

And command groups can be spotted with `{...}` (as in the Proxmark3 client)

Proposals, to be discussed:

* When possible, try to reuse pm3 command syntax.
* Some commands act on current slot only, some require slot argument => make slot argument optional for all
* Group emulator commands with `e*`, e.g. `hf mf e*` (eload,...)
* When a switch to reader/emulator mode is required, do it automatically. Show mode in the prompt.
* (not that the list below doesn't always document the long options, it's just based on a quick automatic dump)
```diff
- hw chipid get
+ hw chipid
- hw address get
+ hw address
- hw mode set                -m {reader,r,emulator,e}
+ hw mode                    -r/-e/--reader/--emulator
- hw mode get
+ hw mode
- hw slot nick set           -s number -st number -n NAME
+ hw slot nick               [-s number] -st number -n NAME
- hw slot nick get           -s number -st number
+ hw slot nick               [-s number] -st number
- hw slot nick delete        -s number -st number
+ hw slot nick               [-s number] -st number -d/--delete
- hw slot list               [-e {0,1}]
+ hw slot list               [-e/--extend]
  hw slot change             -s number
- hw slot type               -s number -t number
+ hw slot type               [-s number] -t number
- hw slot delete             -s number -st number
+ hw slot delete             [-s number] -st number
- hw slot init               -s number -t number
+ hw slot init               [-s number] -t number
- hw slot enable             -s number -st number -e {0,1}
+ hw slot enable             [-s number] -st number
+ hw slot disable            [-s number] -st number
- hw slot update
+ hw slot store               // cf hw settings store
  hw slot openall
- hw ble bonds clear
+ hw settings bleclearbonds   // + warning that it's done without hw settings store
- hw settings animation get
+ hw settings animation
- hw settings animation set  -m {0,1,2}
+ hw settings animation      -m {0,1,2}
- hw settings btnpress get
+ hw settings btnpress
- hw settings btnpress set   [-l] -b {A,B,a,b} -f {0,1,2,3,4}
+ hw settings btnpress       [-l] -b {A,B,a,b} -f {0,1,2,3,4}
  hw settings store
  hw settings reset
  hw settings blekey         [-k KEY]
- hw settings blepair        [-e ENABLE]
+ hw settings blepair        [-e/-d/--enable/--disable]
  hw connect                 [-p PORT]
  hw version
  hw dfu
  hw factory_reset           [--i-know-what-im-doing]
  hw battery
  hw raw                     -c COMMAND [-d DATA] [-t TIMEOUT]
  hf 14a scan
  hf 14a info
  hf 14a raw                 [-a] [-s] [-d DATA] [-b BITS] [-c] [-r] [-cc] [-k] [-t TIMEOUT]
- hf mf nested               [-o] --block-known decimal --type-known {A,B,a,b} --key-known hex [--block-target decimal] [--type-target {A,B,a,b}]
+ hf mf nested               [-o] -k <hex> --blk <dec> [-a(default)/-b] --tblk <dec> [--ta(default)/--tb]
  hf mf darkside
- hf mf rdbl                 -b decimal -t {A,B,a,b} -k hex
+ hf mf rdbl                 --blk <dec> -a/-b -k <hex>
- hf mf wrbl                 -b decimal -t {A,B,a,b} -k hex -d Your block data
+ hf mf wrbl                 --blk <dec> -a/-b -k <hex> -d <hex>
- hf mf eload                -f FILE [-t {bin,hex}]
+ hf mf eload                [-s number] -f FILE [-t {bin,hex}]
- hf mf eread                -f FILE [-t {bin,hex}]
+ hf mf esave                [-s number] -f FILE [-t {bin,hex}]
- hf mf settings             [--gen1a {1,0}] [--gen2 {1,0}] [--coll {1,0}] [--write {0,1,2,3,4}]
+ hf mf econfig              [-s number]
+                            [--enable-gen1a/--disable-gen1a]
+                            [--enable-gen2/--disable-gen2]
+                            [--enable-block0/--disable-block0]
+                            [--write-normal/--write-denied/--write-deceive/--write-shadow]
- hf mf sim                  --uid hex --atqa hex --sak hex [--ats hex]
+ hf mf econfig              [-s number]
+                            [--uid hex]
+                            [--atqa hex]
+                            [--sak hex]
+                            [--ats hex]
- hf mf info
+ hf mf econfig              [-s number]
- hf detection enable        -e {1,0}
+ hf mf econfig              [-s number]
+                            [--enable-log/--disable-log]
- hf detection count
+ hf mf elog                 [-s number]
- hf detection decrypt
+ hf mf elog                 [-s number] --decrypt
  hf mfu rdpg                -p decimal
  hf mfu dump                [-p decimal] [-q decimal] [-f FILE]
- lf em sim set              --id hex
+ lf em econfig              [-s number] --id hex
- lf em sim get
+ lf em econfig              [-s number]
  lf em read
  lf em write                --id hex
```

GameTec-live, 2023-10-06:
adapting the PM3 syntax is a good idea IMO

iceman1001, 2023-10-07:
the pm3 command structure and params isn't unified so don't fall in that pit.

Make sure you lock down so same parameters is the same across the CLI.   And reserve some.  
Find a nomenclature that fits.   PM3 struggle with it. 

Make a guiding document and adapt,  like:
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/cliparser.md

Same thing when it comes to color management, like
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/colors_notes.md

Learn from the mistakes PM3 CLI did and become even better.

doegox, 2023-10-07:
thanks for the tips @iceman1001 !

closed 2023-10-10 by doegox