RfidResearchGroup/ChameleonUltra #166: [idea] Add ID/UID informations in slot list

spp2000, 2023-10-08:
In the slot list it appears only the nick of that slot without any info about the corresponding ID/UID.
By knowing the ID/UID too, the user should have a better awareness of what's inside a slot (=what Chameleon emulates), expecially after using the "clone UID" function associated with the hardware buttons.

doegox, 2023-10-10:
done :)

closed 2023-10-10 by doegox

spp2000, 2023-10-10:
Ok @doegox, thanks! now in the CLI I see both the nick and the ID/UID. I have another proposal :D at least for LF.
I try to explain:

If I load a saved LF card to a slot, the nick automatically updates with the ID (the nick equals the stored ID).

If, instead, I clone an LF card to a slot, the nick remains the one previously stored in that slot, while the ID is the one just cloned. IMHO there's no need to preserve that nick; the clone function could update the nick too with the ID just cloned, so there will be more consistency.

spp2000, 2023-10-11:
> But we can change/add a nickname when cloning, into something like "cloned", so ppl know where its data comes from. (and there is no problem having several identical nicknames)

OK, this could be fine too. The important thing imho is to change the previous nick to avoid confusion. Thanks

doegox, 2023-10-11:
done :)