RfidResearchGroup/ChameleonUltra #173: Implement async messages

doegox, 2023-10-17:
Current communication is purely synchrone: the client sends a command and the firmware answers.

We'd like to allow async messages to implement the following features:

* from fw to client
  * allow to send debug messages
  * allow to send progress info from long running commands, such as `darkside`
  * allow to send critical battery level alerts
  * allow to send button press events to the client

* from client to fw
  * allow to interrupt a long running command, such as `darkside`, from the client
 
Maybe the implementations changes should be done by taking into account a move to asyncio implementations of serial, BLE and TCP communications in the Python client.

doegox, 2023-10-17:
This will require new STATUS and possibly new dummy Command code
* Command NOP = 0
* STATUS_ASYNC_DEBUG = 0x100 for debug messages (with the NOP Command code)
* STATUS_ASYNC_NRFLOG = 0x101 if we manage to stream NRF_LOG messages (with the NOP Command code)
* STATUS_ASYNC_BATTERY = 0x102 for battery events (with the NOP Command code)
* STATUS_ASYNC_BUTTON = 0x103 for button events (with the NOP Command code)
* STATUS_ASYNC_PROGRESS = 0x104 for progress messages (with the running Command code)

xianglin1998, 2023-10-18:
Some design ideas for this enhancement:

1. The data in the receiving buffer is now directly passed to the command function for execution. If need to implement commands similar to canceling the execution of Darkside, maybe need a double buffer, which means we cannot destroy the previous data.

2. Distinguish between synchronous commands and asynchronous broadcasts using status codes or command codes.

3. PYTHON CLI already has the ability to handle asynchronous commands, but now it is using implementation functions such as blocking and timeout.