RfidResearchGroup/ChameleonUltra #175: Issues with LF emulation or original ultra hardware, no issues on clone

m-kozlowski, 2023-10-18:
I'm having issues emulating em4100 using both of my CUs with reader in my residential block.
Unfortunately I don't know the brand of the reader.
The success rate is about 1/3, but most attempts ends with the RF field diode flashing irregularly and the door remain locked.

It happens on any version of firmware I've tried since August, BUT it worked quite reliably with firmware built using lf_tag_em.[hc] from June "factory release".
No problem when trying to read both CUs using flipper zero or some cheap chinese reader.
CU's taobao clone (the one with triangles on antenna pcb) doesn't seem affected at all, works with problematic reader every time on any firmware.

doegox, 2023-10-18:
electrical contact to the LF antenna via the screws has been reported to be problematic on a few devices, maybe you could check. But be cautious, it has also be reported that stuck screws broken when forced...

m-kozlowski, 2023-10-18:
Yeah, I forgot to mention... I did check that too. The screws was indeed stuck, but no signs of resin residues on connecting surfaces.

xianglin1998, 2023-10-20:
The developers of CU's Taobao clone have submitted repair code for some issues with LF, which I think is the reason why Taobao clone LF can work very well.

xianglin1998, 2023-10-20:
If this continues, I will consider restoring the code to its original state because too many people are complaining about this issue.

panayotoff, 2023-10-23:
I have DevKit and Lite from the original campaign and both cannot emulate LF ( tried with flipper, proxmark, door acess ... no ). The ultra does emulate and the DevKit can also read LF, so I don't think it's hardware issue in my case, but it is very annoying issue.

Ultimator14, 2023-11-03:
Same here. LF emulation is broken for me on ultra, firmware 2.0.0. Can't read with either flipper or proxmark.   
LF reading works (via the app). For HF both works.

ShigemoriHakura, 2023-11-03:
> The developers of CU's Taobao clone have submitted repair code for some issues with LF, which I think is the reason why Taobao clone LF can work very well.

Actually I think it's not the software issue but a hardware issue. (Both cu can install the latest software). original cu has about 170uH, but the "clone" version has about 373uH which acts more like a real "id tag". 
BTW plz split me and the "Taobao clone", I have never sold or contact with Taobao sellers from first to last.

nieldk, 2023-11-22:
FWIW i havent experienced this issue on my devices. Maybe some changes In HW revisions (mine was In the first badges shipped)

ffcrus, 2023-11-25:
I have the same problem. My Chameleon Ultra can't emulate lf cards. I tried using my Chameleon Ultra from a mobile application (even two different mobile applications). In the Iceman`s discord I was given a recommendation to try using CLI. And so, I installed CLI and nothing changed. CU emulates only hf cards. At the same time, it reads EmMarine cards perfectly. What I tried to do:
- Update firmware. The problem was reproduced on all three firmwares.
- Configure the slot from the official Android application
- Configure the slot from an unofficial application for Android
- Reset Chameleon Ultra to factory defaults from the mobile application (only on the latest firmware)
- Reset Chameleon Ultra from CLI to factory defaults
- Readed the card and tried to emulate it
- Emulate the card that comes after a factory reset
- Disable all slots except one
- Disassemble and clean the antenna contacts with alcohol. If you unscrew one screw from the contact, the Chameleon Ultra loses the ability to read EmMarin cards, which is logical. After cleaning and screwing in, readability is restored, but emulation does not work.
- Each time I tried to read with a regular reader, Proxmark3 and Flipper Zero. None of the methods could read the lf card emulation even once. On the Chameleon Ultra, the white LED sometimes blinks, but emulation does not occur. What should I do in my situation?

panayotoff, 2023-12-01:
> On the Devkit and Lite LF emulation works fine, on my Ultra it doesnt. (Tried reading with the pm3 or Flipper zero, nothin) Rolled back to v1.0, works perfectly fine. If someone can compile a current version without the commit in question, im happy to try that too, but i think rolling back the commit might be advisable. @xianglin1998

I have the opposite problem, on Ultra the LF emulation works fine and on my Lite/Devkit it does not. Devkit is able to read LF just fine though, just cannot emulate them back...and I've tried.

m-kozlowski, 2023-12-27:
I'm trying to understand differences between v.1.0 (a.k.a. "factory version") and current LF emulation code.
If I get it right, in the old code `lf_is_field_exists()` is called *after* transmitting whole ID LF_125KHZ_BROADCAST_MAX times,
while in the current version `lf_is_field_exists()` is called "_during no-mod half of last bit of last broadcast_"

There is 20us delay in `lf_is_field_exists()` function. Maybe this messes up transmission somehow? (that's 2.5x cycle interval at 125kHz)

nieldk, 2023-12-28:
> The developers of CU's Taobao clone have submitted repair code for some issues with LF, which I think is the reason why Taobao clone LF can work very well.

Can you point to that change?

nieldk, 2023-12-28:
Looks like m_is_send_reboardcast_last_edge  is unititialzed in firmware/application/src/rfid/nfctag/lf/lf_tag_em.c, Line 46 (from commit d4e6c03) it probably should be initialzed to false.

GameTec-live, 2023-12-28:
Did some testing, no change. This seems very inconsistent nomatter if i leave it as be, set it to true or false or whatever...
Just for reference, pm3 log, you cant tell where i switched fw...
[log_20231228190253.txt](https://github.com/RfidResearchGroup/ChameleonUltra/files/13788981/log_20231228190253.txt)

enselerizer, 2023-12-28:
I have the same problem. I conducted experiments with some noname reader that I have, and noticed that the Chameleon Ultra needs to be brought much closer to the reader than the original tag in order for it to be reliably read. If you move it too far, the RFID activity LED begins to flicker and reading does not occur. Perhaps this is a design issue with the LF antenna?

xianglin1998, 2024-01-17:
> I have the same problem. I conducted experiments with some noname reader that I have, and noticed that the Chameleon Ultra needs to be brought much closer to the reader than the original tag in order for it to be reliably read. If you move it too far, the RFID activity LED begins to flicker and reading does not occur. Perhaps this is a design issue with the LF antenna?
> 
> P.S. I tried all versions of the firmware, including the "factory default" version, and the problem occurs on all of them. I made sure several times that the contact between the back cover and the nuts on the front cover was good. My “problematic” reader has weak field strength compared to a normally functioning reader (tested using Proxgrind RFID Field Detector).

Due to the impact of components and PCB layout on RFID radio frequency, ChameleonUltra will inevitably reduce the card reading distance in terms of Tiny size. If excellent RFID performance is required, you should try using ChameleonLite or Chameleon Ultra Dev

xianglin1998, 2024-01-17:
@GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

xianglin1998, 2024-01-17:
Artifacts from lf_test_rollback branch.

xianglin1998, 2024-01-17:
Work well on my reader, so i need you help, to test this change.

xianglin1998, 2024-01-18:
> @xianglin1998 trying to do a before test, it seems to have magically started working for me... my pm3 can read it perfectly and it isnt even very position dependent... So, ig i cant help anymore, sorry...

Let's look forward to testing by other users.

stereolacing0z, 2024-01-18:
> @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

Well, I have this problem, and it is definitely something floating and really embarrassing.
I have two Ultra and two Lite: conducting tests by just placing one over the other and doing reading via an app can confirm I have a problem with all of them. The same behavior is reproduced when trying to read chameleons by VIZIT entrance panel reader (I don't know the exact model), while there is absolutely no problem reading them by a handheld reader (5YOA IDICW01).

While RF LED lights up even 2-3 cm away, LF requires an extremely short distance of 1-2mm between ultras to actually work, and even in these conditions, being vise-fixed, it sometimes stops working after 1-2 reads (RF flashes) or works inconsistently, getting data once per 2-3 reads. It does not work at all with VIZIT panel (just flashing led once when entering the field).

Reading a regular LF tag works from a much higher distance, over several centimeters, up to the same distance as RF LED on emulator catches the request.

Absolutely no problem with HF, though.

Problem is constantly reproduced on all devices with 2.0 builds I've tried, including this build. Didn't test with 1.0.

Feels like it may be a problem with transmitting power or transmitting **timing**.

Yzord, 2024-01-23:
> @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

How can i install this? I am on v2.0, but i guess this is a new (test) firmware release?

whyislow, 2024-02-19:
> > @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts
> 
> How can i install this? I am on v2.0, but i guess this is a new (test) firmware release?

thnx, it's work, 10/10 
before this update i can't do anything with my lf, sometimes... 1 in 100

whyislow, 2024-02-19:
> @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

thnx, it's work, 10/10
before this modify can't do anything with my lf, sometimes... 1 in 100

xianglin1998, 2024-02-20:
> > @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts
> 
> How can i install this? I am on v2.0, but i guess this is a new (test) firmware release?

You can use NRF DFU Application to update this firmware, or try to use ChameleonUltraGui to update(I dont known it is support or no, you need to confirm)

ffcrus, 2024-03-12:
> @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

Nothing changed for me. Still doesn`t work LF emulation.

R3zdrake, 2024-03-25:
Hi all, just wanted to report, i have similar issue where my CU was unable to read any LF cards. I have followed the troubleshooting steps and i do not found any epoxy residue. I can confirmed that the CU LF reading is working fine when i have tested with basic factory firmware and it was able to read the LF card however when i install the lastest firmware, i'm no longer able to read the LF card again. 

I have also tried the modified version below : https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

It is not working either. 

[image]
[image]

Kuninoto, 2024-04-24:
Hi all, FWIW my CU just arrived, I've updated its firmware to v2 and tried the LF Read+Emulation and it worked wonderful.

stereolacing0z, 2024-04-24:
Well, after a significant number of different tests, I feel like something is drifting away in time when using CU or Lite with specific readers.

You'll be able to emulate the card once or even several times after the firmware reset, but then it stops emulating correctly. I'm not sure about just replacing the LF ID or enabling/disabling the slot: it gives uncertain results.

Meanwhile, with other readers, especially handheld Chinese copying devices, it works fine every time.

Maybe there is a chance of some data shifting over time in the sequence to be emitted? Kinda, we emit a stop-byte in the middle, but the handheld reader ignores the pause in the middle of the sequence? I am not very familiar with RFID protocols.
Also, it may be a timing issue, but I have no idea how to test all of that.

Nextery, 2024-05-12:
> @GameTec-live Try this modify: https://github.com/RfidResearchGroup/ChameleonUltra/actions/runs/7554329636#artifacts

Wanted to try, but it's expired :(

DemetryP, 2025-02-05:
I had the same problem until I tried bringing the tag (or reader) not to the back of the case, but to the front, which has LEDs.  If you do this, everything works fine (version 2.0)

Snollyg0ster, 2025-05-16:
> I had the same problem until I tried bringing the tag (or reader) not to the back of the case, but to the front, which has LEDs.  If you do this, everything works fine (Chameleon ULTRA SE3, version 2.0)

Thank, it works for me

ndguardian, 2025-07-02:
I just wanted to pop in here and mention I've seen this issue as well. I have a Chameleon Ultra. On factory firmware 1.0, LF reads work. On 2.0, dev, etc...it simply says no card found. That tells me the hardware is good but something isn't playing nicely on 2.0-based firmware. I've also been able to repeat this, with downgrading, reading card, upgrading and no longer being able to.

I also tried using both the side with the buttons vs the side with the LEDs, and it makes no difference on my end. In pretty much all of these tests, the Chameleon is sitting directly on top of the card too, so I'm confident it's not a distance issue.

I see there's a debug mode in the GUI app, but based on the message I saw, that's probably not what I need to actually debug? And I've also tested with the CLI application, but no dice there too. :/

W1BTR, 2025-12-29:
Thought Id give this a bump and offer any testing needed. Any logs I could pass on or I could send over a tag for testing?

dragoscoca, 2026-01-05:
Same problem here. Emulation on LF doesn’t work. 

Bump!

hammcmod, 2026-01-05:
Sam problem here. It beeps but fails. I plan to read it with an hid proxmark to seenhow it varies

jrvalino, 2026-01-11:
Same in here. I have bought also the clone. I can save the lf and place it to 1 of the 8 lf tags but also not working. It just light up the led but can't emulate

simplumd, 2026-01-14:
Same here woth the clone from aliexpress, emulation doesn't work

lef1337, 2026-01-19:
Same here 2/3 times it doesn't work on Lf

ferdhagethorn, 2026-01-22:
(snipped for brevity)
> Same issue with emulation on my clone. software - timings of emulation are not right
> 
> Since the CU reads LF cards fast and accurately, it seems the antenna is working fine... Reading and emulating do use the same antenna, but the RF fields are different - will experiment with the physical side some more.

In my case the poor LF performance was the antenna - physical issue... what worked for me; The clones use some sort of glue on the screws, and when these aren't tightened down well enough before the gluing process the glue becomes an insulator as it seeps in between the screw and the hole through the antenna board. 

I scraped the goo off, removed the screws, removed the glue from the bottom of the screws (2 of 4 had glue seep underneath) and made sure the antenna board was clean by wiping it with a bit of isopropyl-alcohol. Tighten the screws gently, as you can overtighten and then they keep spinning.

Now my flipper zero scans LF tokens emulated by the CU in about 2 seconds, much less position sensitive, even hovering 1 cm above it. 
The antenna must face the back of the flipper zero, as it has an RF shield on the back.

voltlog, 2026-01-22:
@ferdhagethorn which version of hw & fw are you running? I have a clone with hw ver 2.0, fw ver 2.1 and LF read works fine, emulation doesn't work. I have taken the device apart, there is no glue on the screws or anything else that might prevent a good connection. Reading LF tags is not at all position sensitive, signal seems to be very good.

deckstarrnz, 2026-01-22:
Interesting piece of info in this video https://www.youtube.com/watch?v=O0VUQ5uKagk, the original has two springs... looks like it's on the opposite side of the buttons (bottom two on the image above)... those springs are missing on my clone and that might be the issue.

deckstarrnz, 2026-01-22:
When trying to emulate the hid prox, the white light just blinks, no scan is happening. Seems the readers I'm trying on are pulsing the induction loop causing the CU to restart its send. It blinks every second. When reading the hid prox with the Flipper Zero the light stays on.

brewt, 2026-02-07:
> I have the same problem with my aliexpress clone.
> 
> 1. I can read LF tags
> 2. I can write LF tags
> 3. When i used my Proxmark in emulation mode and bring the Chameleon close to it, the "WIFI" LEd blinks, but no tag is read
> 4. One time Proxmark was able to read the tag

I'm curious what the differences are between all the various AliExpress CU clones are.  What's the tag configuration you're emulating on your CU?  I also have a AliExpress clone as well as a proxmark3, so I can test it as well.  This could help narrow down where to look for further debugging.

ITCMD, 2026-03-09:
> > [@damtrx](https://github.com/damtrx) a PCB can contain DNP (Do not populate) components and still work fine, if the PCB designer has decided that. Clearly you are looking at two slightly different PCB designs so visually missing components between the two is not directly a good reason for the LF emulation feature working or not.
> > **There is also no mention of the firmware you are running on these.**
> > I have the same clear plastic v2.0 hw but immediately after receiving it I flashed it with fw ver 2.1 and my LF emulation doesn't work. I did not test LF emulation with the original fw it came with.
> 
> I'm not sure how many PCBs you have designed in your life but a proper design doesn't have the option of "do not populate" Usually in this case someone decided to cut corners. Most of the times it is from stabilizers, caps, resistors, diodes.
> 
> So the problem from my point of view is that someone decided to cut a few cents.
> 
> BTW I do pcbs as a hobby. Not at this level but it is clear to me that there is a problem

I've worked on many projects that have "do not populate" on the official design. I think this is less standard on commercial products, but on hobby projects it is common to leave a "do not populate" when a component is not needed or a change is made if re-designing to board isn't _necessary,_ if it makes no difference to save time or to allow testing in the future.

mfcarroll, 2026-10-23:
@m-kozlowski I've been doing some work on lf emulation recently that I suspect might relate to some of the issues in this thread. Cases where nothing at all will read the emulation are almost certainly a hardware issue (missing springs makes sense, as those carry the connection to the antenna plate). But in your original case specifically - where flipper can read it, but *not* your building wall reader, I suspect that could be a result of a clock frequency mismatch between the CU and the wall reader field.

With hardware tags, the frequency of the field doesn't matter at all, because the tag's frequency comes from the reader. With emulation on the CU, it's the CU's own crystal oscillator that generates the rf. So a mismatch between the two could cause the reader to reject the emulation, or make it temperamental.

Flipper has the ability to tap the raw 125kHz signal and then locks its emulation to that, so the emulation stays in lock-step with the reader. From what I can tell, the low level hardware in the CU doesn't allow for that. (If I'm right about that, it would be an obvious improvement for a future board revision.)

If you're still at the same building, there are some thing you can check that would help narrow down if that's the case. I'd be really interested to work with you on it. Until now I've had a theory that frequency mismatch could be a real-world issue with certain readers and certain emulation types, but no way to prove it, as the Indala wall reader I have access to appears to be tolerant of the frequency mismatch with my CU's.

First, what credential type is it?

Second, is the emulation from your flipper accepted by the wall reader?

Third, have you measured the reader's frequency? If not, you can using the flipper. (It's a passive measurement, not a probe, and should be no risk even on a real system.) Go to Apps - Tools - NFC/RFID Detector, and hold the flipper up to the wall reader. For an on-spec reader it would list exactly 125.00.

I could also be completely wrong here and frequency mismatch might be nothing to do with it. It's a hunch I'd be interested to at least check / rule out if possible. It's interesting that you mentioned an older version of the firmware worked. I'll take a look at that.

m-kozlowski, 2026-10-23:
sounds like a pretty plausible explanation, actually :)

unfortunately, since i reported the issue, the building management has replaced the whole access control system with a MIFARE-based one, so i can't really test it anymore.

mfcarroll, 2026-10-24:
Thanks. That's too bad they changed the system. Most likely they got rid of the old readers but if you happen to be friendly with the management you could ask if they have one still in a box somewhere. Getting hold of real readers to test on seems harder than I'd imagined it would be. I'll look at that old firmware anyway when I get a chance. There might be a clue in there. Do you happen to know which commit / version it was exactly that worked for you previously. And do you still have a copy or know exactly what type of credential it was?

mfcarroll, 2026-11-09:
Follow up on my clock mismatch comment above - I've found another possible cause for this kind of problem on a real reader. A Paradox door reader that rejected the CU. It turns out it pulses the field. It switched on for about 50 ms, then off, repeatedly, instead of keeping the field on. The CU has to start sending a complete frame very quickly inside each pulse. That reader was also about 0.5% off 125 kHz. Details are in #532.

@m-kozlowski the "RF field diode flashing irregularly" in your original report might well have been the same thing, though there's no way to tell now that the reader is gone. Just noting that in case anyone else comes across the same behaviour.

If anyone else has a reader that a T5577 opens but the CU doesn't, I'd be interested in what the reader itself does exactly. This page flashes experimental firmware on the CU that measures a reader's field (frequency, whether it pulses, and the timing), and saves the result so you can read it back afterwards. It doesn't record your card's credential.

https://mfcarroll.github.io/rfid-test-tools/chameleon/field/

If you have a Flipper, you can also try measuring the field with the updated rfid/nfc detector app here. Calibrate first on a computer, then use the app at the reader.

https://mfcarroll.github.io/rfid-test-tools/flipper/crystal/

(Do both if possible - the CU experimental firmware now measures some things the Flipper does not, such as looking for HiTag2 activation commands, but the Flipper gets a more precise measurement on a field that stays on. The Flipper can count the field directly and aggregate that over time, and the CU test uses a fast fourier transform to find the frequency since the CU isn't able to tap the field directly as far as I can tell.)

mfcarroll, 2026-11-12:
A bit more on the clock mismatch idea. It only applies to PSK credentials (currently IDTECK, in main since #407, and Indala in #509). EM410x, HID and the other ASK/FSK types don't care about a clock offset this size, so this doesn't explain most of the reports in this thread.

My pm3 runs about 104 ppm slow and my CUs run about 36 ppm fast, so about 140 ppm apart in total. With the CU emulating IDTECK on main, `lf read -s 8192; lf idteck demod` decodes 0 of 40 reads. On an experimental build that measures the reader's clock and trims the CU's subcarrier to match it (the self-trim mentioned in #532), the same CU decodes 34–36 of 40, and a second CU decodes 37 of 40.

#407 argues that the CU doesn't need to lock its subcarrier to the reader's clock, because a PSK1 reader decodes the phase change between one bit and the next. That holds even at 140 ppm: the phase shifts less than 1° per bit. A reader that compares each bit with the one before it won't notice, which fits #407's test on a real IDTECK reader.

The pm3 works differently. It takes exactly two samples per cycle of the 62.5 kHz subcarrier, so a clock offset makes the signal it sees fade in and out, about every 57 ms here. A read that lands on a fade fails. The fades start at the same point each time the field comes on, so every ordinary `lf read` lands on the same fade. That's why main reads 0 of 40. With the field held on for 1 s, main decodes about 30 % of the time, between the fades.

That doesn't contradict the pm3 test in #407's review. The pm3 reads the CU fine when the two clocks are close. On my tests it reads every time within about ±15 ppm and gets gradually worse beyond that. My devices are just far apart.

The Flipper is in between: it reads the CU within about ±40 ppm and never beyond about ±55 ppm. So whether the offset matters depends on how the reader demodulates, and how far its clock is from the CU's.

I still haven't found a door reader that rejects the CU for this reason. My Indala readers read CU emulation without the trim. I figured it was worth writing up the explanation in case one comes along, either for IDTECK/Indala or one of the other PSK types as and when those are added to the CU (e.g. Keri or NexWatch, which I have experimental support for, not upstreamed yet).

If you have a PSK reader (IDTECK, Indala) that a T5577 clone opens but the CU doesn't, the reader's frequency would show whether this could be the cause. The Flipper page linked above can be used to measure it.