RfidResearchGroup/ChameleonUltra #177: PSA: Silicone case dissolves the epoxy

osulivanDE, 2023-10-20:
Not a real bug but still bugs me: The silicone case dissolves the beautiful epoxy layer and gives it pimples
[image]

GameTec-live, 2023-10-20:
ig me destroying the case while putting the CU in was a happy accidemt afterall lol
but thats wierd, never heard of silicone just through contact dissolving epoxy...

llliii0325, 2023-10-23:
Can you share the cleat production documents between the silicone casing and the upper and lower pcb?

ekoyle, 2023-10-29:
This could also be caused by improperly cured epoxy (wrong amount of catalyst, not mixed well enough, etc), so it may be an issue with the process rather than a reaction with the silicone cover.

Msprg, 2023-11-19:
Just today I removed my Ultra from the silicone case for a while and noticed some unevenness on the epoxy. Shrugged it off as "always been there" but this makes more sense.

That said... What the actual heck!

sfjuocekr, 2024-07-17:
It is most likely due to plasticisers in the silicone!