RfidResearchGroup/ChameleonUltra #183: Staticnested results quite poor compared to pm3 implem

doegox, 2023-11-11:
See https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/mfkey/staticnested.c and https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/mfkey/test_static.sh

pm3 implem can break the 9 samples while Ultra implem can only break 2 of them...

xianglin1998, 2023-11-11:
Let me check it.

 
Assigned #183 to @xianglin1998.

xianglin1998, 2023-11-15:
Bug Fixed.

xianglin1998, 2023-11-15:
For this problem, explain:
Cause by 'uint64_t' - 'uint64_t' maybe overflow return value.
Chinese: https://blog.csdn.net/ryfdizuo/article/details/49872025
English: https://stackoverflow.com/questions/36681906/c-qsort-doesnt-seem-to-work-with-unsigned-long

xianglin1998, 2023-11-15:
Mfoc have the same problem: https://github.com/nfc-tools/mfoc/blob/ba072f16f6b2a655d51da2171ecfb83e26c0ef58/src/mfoc.c#L1199

doegox, 2023-11-15:
Fixed, thanks https://github.com/nfc-tools/mfoc/commit/a5a4c9184d7b2a92c9a2f507009319ea9747bb30

xianglin1998, 2023-11-15:
Perfect, wait me from busy to free, hahaha. continue !!!