RfidResearchGroup/ChameleonUltra #185: Support for Pyramid cards?

victorhooi, 2023-11-20:
Is there any chance of adding support for Pyramid tags in Chameleon?

For example, I have some Pyramid tags - I can read them successfully with a Proxmark RDV4:

```
[usb] pm3 --> lf pyramid reader
[+] Pyramid - len: 81 -unknown- Card: 21134, Raw: 000101375b0101010101016e34XXXXXX
[usb] pm3 --> lf pyramid demod
[+] Pyramid - len: 81 -unknown- Card: 21134, Raw: 000101375b0101010101016e34XXXXXX
[usb] pm3 --> lf search
​
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=] 
[=] Checking for known tags...
[=] 
[!] ⚠️  Specify one authentication mode
[+] Pyramid - len: 81 -unknown- Card: 21134, Raw: 000101375b0101010101016e34XXXXXX
​
[+] Valid Pyramid ID found!
​
[=] Couldn't identify a chipset
```

However, if I try to read it with the LF read on Chameleon Ultra, it says no tag found.

wrenchii, 2023-12-13:
Yes please! Facing the same problem on a pentest (easily read and emulated on RDV4 and FZ). While I got the FZ to do the task (and / or a T5577), I would ideally want to consolidate everything into the Chameleon Ultra where I can.

Spider624, 2024-08-07:
still no info and no support for this