RfidResearchGroup/ChameleonUltra #192: The GUI don't save new card

Tozneno, 2023-12-25:
With Android app, When i want to save a new card i've juste reading, the GUI ask me a name for it then i try to press ok button but nothing append.
I can save it to .bin form but not directly in the application.

I precise that the first version of the GUI app i used work right for saving.

GameTec-live, 2023-12-26:
it works for me. Are you sure it doesn't create a new card under saved cards?

Tozneno, 2023-12-26:
Nop i try to save with différents name, i also reset my chameleon ultra for re testing.
No New card added 🙁

GameTec-live, 2023-12-26:
what about some more info like os, etc

Tozneno, 2023-12-26:
Ok for now i try first Time to un install and reinstall the app, same problem.
But i try to delete app data storage in Android app manager, and it works again now.
Very strange...
When i re install the app it say to me some permission problem. Let see with a pict
[image]

GameTec-live, 2023-12-26:
did you forget to grant it the permissions required for bluetooth?

Tozneno, 2023-12-26:
I grant every permissions the gui app Ask me.
The error message don't ocure anymore After this single appear.
For resuming some wizardery append but for now everything is ok again with the gui app.

GameTec-live, 2023-12-26:
? so everything works now or?

Tozneno, 2023-12-26:
Yep 👍

GameTec-live, 2023-12-26:
mind closing this issue then?

closed 2023-12-26 by Tozneno

Elinas-of-Alba, 2024-01-21:
No, have the same issue on Mac OS GUI and iOS. If I want to save the card it does simply nothing.