RfidResearchGroup/ChameleonUltra #196: BLE pairing to iPhone does not work

ponas-niekas, 2024-01-04:
I can't seam to pair CU to my iPhone 13..

1. I open Chameleon Ultra GUI on the iPhone
2. On the home page I see "Connect"
3. I click A or B CU buttons to wake it up
4. I click on "refresh" button in Chameleon Ultra GUI
5. Spinning wheel icon shows up and then disappears few seconds later. 
6. CU does not appear

With MTools:

1. I click on bt icon
2. A windows pops up with "Filter Devices" "on"
3. I click A or B CU buttons to wake it up
4. and... nothing happens, CU does not appear :/.

CU is running latest firmware 2.0.0-128-gfefcde5. 
Opening "Device Settings" of Chameleon Ultra GUI running in Ubuntu, BLE pairing is "Enabled", pin is 123456.
Am I missing smth ?

GameTec-live, 2024-01-04:
hm, weird... disabling the pin fixes it or?

ponas-niekas, 2024-01-04:
Disabling the pin did not help. I also tried resetting the settings and factory reset. And installed Chameleon Ultra GUI on another iPhone 13 with the same result..

GameTec-live, 2024-01-04:
but usb works? (eg on desktop)
and you did give it permission?

ponas-niekas, 2024-01-04:
USB works fine, yes. And Chameleon Ultra GUI and MTools apps do have bluetooth (if that's what you are asking) permissions on iPhone.

GameTec-live, 2024-01-04:
does it show up under the ios bluetooth settings?

ponas-niekas, 2024-01-04:
No, it does not, that's the issue..

GameTec-live, 2024-01-04:
> No, it does not, that's the issue..

it should show up but youll be unable to connect (only possible inapp)...

tried reflashing fw?
as it also doesn't work with the pin disabled then, you might have a dead BLE subsystem / hardware fault in that region...

ponas-niekas, 2024-01-04:
I "re-flashed" firmware to the latest version today.. Anything else to try ?

GameTec-live, 2024-01-04:
you can check if it shows up with bluetooth when connected to a pc via usb (then its awake too)
other than that, no clue, sorry...

ponas-niekas, 2024-01-04:
Checked that too, but still no sign of BLE, faulty hw I guess. I contacted the seller.

rawGdawG, 2024-01-08:
I'm having the same issue. No bluetooth discoverability of my device. It goes from awake to sleep within 3 seconds. It can be connected with a USB, but even after enabling BLE pairing, changing the PIN, factory reset. Still no bluetooth discoverability of the device.

Yzord, 2024-01-13:
Hmm, i guess i have the same thing. I was googling how to connect to BT on iphone and stumbled upon this topic. When i push A or B button a red light goes on, goes like KITT from Nightrider and nothing happens. Can’t find it in any app available atm.

GameTec-live, 2024-01-13:
you should still be able to connect with the old firmware and CUGUI then forces you to update, but atleast it works for you now @Yzord

meditant, 2024-03-16:
Same problem, work before with iPhone and Mac with BT. i Upgraded ChameleonUltra firmware and now unable to discover device, i reset firmware, pin and nothing change, unable to discover the Chameleon Ultra.

dittohead, 2025-01-24:
I have aliexpress clone, and same issue. If press nrf chip-it start to work over bluetooth. Blow a little hot air from heatgun should help

octaprog7, 2025-01-24:
My cell phone won't connect to the device via Bluetooth if GPS is turned off.

dittohead, 2025-01-27:
> I have aliexpress clone, and same issue. If press nrf chip-it start to work over bluetooth. Blow a little hot air from heatgun should help

Today I back to my lab, and try to put some hot air to NRF, but problem with china clone more complicated:
I put PCB to heatbed and start to desolder NRF ic, and found that IC's pin broken from IC case:

[image]

nico0481, 2025-07-29:
Same issue on Android with CU.

Works fine over bluetooth with MToolsBLE, but works only via USB-C in ChameleonUltraGUI.
No CU device discovered.
Latest firmaware