RfidResearchGroup/ChameleonUltra #198: Unable to emulate mifare classic 1k

mindbreaker2, 2024-01-15:
I tried to emulate a mifare classic 1k and it did not work. I've checked the dumps and they are identical.

The chameleon ultra is up to date and sending the correct data so I don't get it.

xianglin1998, 2024-03-07:
Update your device firmware.

mindbreaker2, 2024-03-09:
I've updated it but still no luck. The flipper zero does not work either so I think my reader is doing something to be sure it's a real tag. 

As soon as the cu communicates with the reader, a red solid led turns on on the reader.

Nikotine1, 2024-11-30:
I'm having the same issue. The CU has been able to emulate anything I wanted it to until now (both HF and LF). I installed an NFC lock for my mailbox and it works with the HF tags it came with. I took a dump of the tag and confirmed the dumps are exactly the same with NFC tool diff function. Yet, the CU doesn't work on the lock. I can't image such a simple lock has checks for cloned tags, but if it has, which options should I enable (Gen1A magic mode, Gen2 magic mode, Use UID/SAK/ATQA from 0 block or any write mode)?

mindbreaker2, 2024-11-30:
Nothing's wrong with the emulation, the issue is that some readers use non standard protocol. 
I've tried every mode of the cu nothing seems to be working with my reader.

I tried sniffing the traffic to understand the difference between real and emulated tag but I don't have enough knowledge to figure it out.

Nikotine1, 2024-11-30:
Hmm strange. FYI, it's a Sporgo NFC lock ([Amazon](https://www.amazon.com.be/-/en/dp/B0CL9WYDCQ?ref=ppx_yo2ov_dt_b_fed_asin_title)).