RfidResearchGroup/ChameleonUltra #200: [idea] Fido2 over nfc or usb

javierckr, 2024-01-23:
Could be possible to implement fido2 or u2f in the chameleon ultra hardware? or it has any hardware limitation?

javierckr, 2024-01-23:
https://github.com/makerdiary/nrf52-u2f is this the same nrf52 as chameleon ultra?

GameTec-live, 2024-01-23:
you could, but on the other hand theres the question of if you should...
There are more secure and specialized devices like the Yubikey which are better fit for the job...

javierckr, 2024-01-24:
I think it could be a nice feature for developing and testing, I'm trying to flash that firmware over dfu but it's failing.

GameTec-live, 2024-01-24:
for developing and testing? what does this have to do with developing and testing other than get in the way lol

GameTec-live, 2024-01-24:
btw: maybe you're failing to flash because it isn't compiled for the right exact processor? id check that first...
or you could also have the wrong dfu package...

cylin577, 2026-10-31:
> 5\. for developing and testing? what does this have to do with developing and testing other than it getting in the way lol

I mean you can have a smol device for NFC/RFID AND U2f/FIDO2 research/develop device that you have all components of it could be really cool