RfidResearchGroup/ChameleonUltra #202: [idea] LibNFC

abjugard, 2024-02-01:
When connected to a PC via USB, it would be amazing if it could be used via LibNFC to read/write tags.

Is this possible or would the hardware need to be different for this to work?

xianglin1998, 2024-02-20:
Can let CU to emulate a PN532, all usb serial port data translation to pn532 register one by one.

xianglin1998, 2024-02-20:
Like ACR122U.

xianglin1998, 2024-02-20:
But this work is very massive and strange(The mfcuk and mfoc is very slow, why use it)

augustozanellato, 2024-02-22:
> Like ACR122U.

For reference: ACR122U actually has a pn532 inside, libnfc just uses some passthrough commands that bypass the MCU logic and send commands directly to the pn532

abjugard, 2024-02-23:
> Can let CU to emulate a PN532, all usb serial port data translation to pn532 register one by one.

Or we could do that on the computer side through a driver, wouldn't that make more sense?

doegox, 2024-02-23:
it is none-sense to try mimicking a 20 year old proprietary TAMA protocol just to be able to reuse some old and inefficient attack implementations.
What could be considered is to implement some ifd driver to be able to use the Ultra as a basic PCSC reader and be able to use compatible softs like ePassport and EMV readers at the ISO14443-4 layer.

NFC-funs, 2024-03-07:
> When connected to a PC via USB it would be amazing if the Chameleon Ultra could be used with LibNFC-based tools like mfoc, mfcuk, etc.
> 
> Is this possible or would the hardware need to be different for this to work?

Do you find PN532Killer? PN532Killer is a PN532 emulator, a NFC Emulator,and a Sniffer same as Chameleon Ultra.

PN532Killer can be used with [MifareOneTool](https://github.com/xcicode/MifareOneTool), it's a Gui software based libNFC.

LynxZeta, 2024-10-20:
CU actually has a Ci522 inside. 
The Ci522 is pin to pin compatible with RC522.
The PN532 supports more protocols than RC522, including NFC capabilities‌. But most of the features are the same. 
From this chain it seems possible, but it involves a lot of development I think.