RfidResearchGroup/ChameleonUltra #203: Electra intercom

andreivinca, 2024-02-16:
In Romania, there is a well-known brand of intercoms called Electra. The Flipper can detect its tag as EM4100, but the intercom does not respond to its emulation. In other words, the intercom does not react at all to the EM4100 tag, and I've attempted using various tags from the same brand without success.

Another discussion involves individuals who have managed to make Electra work with Flipper.
https://forum.flipper.net/t/electra-intercom/6368

Can this be applied to ChameleonUltra?

GameTec-live, 2024-02-16:
Did you try with the CU?
It does support EM401X

andreivinca, 2024-02-16:
Yes, that was the first step I took.

doegox, 2024-02-16:
the proxmark3 supports it, not yet the cu.
Electra format is a bit similar to EM4100 but with extra data.
https://github.com/RfidResearchGroup/proxmark3/commit/ee22a824ce6ee0048245298b2ebba0be5fb02b6c

iceman1001, 2024-02-16:
There is a more modern electra format which is discussed in that link.    
The PM3 doesn't not identify that part yet.    

I did add the cloning support.
https://github.com/RfidResearchGroup/proxmark3/commit/c544d67d2bc7d49579da4e197ee50fa85a37343b

andreivinca, 2024-02-17:
Could you work your magic for the CU as well?

imeSven, 2024-02-26:
Hey. I got same problem with electras. Unfortunately my home and my work uses same intercom and I bought the CU specially for those to get rid of the tags 😂. I really hope it Will work soon.

yuretszzz, 2025-01-31:
Up. My be it can be cloned manually via terminal?

sandumarius, 2025-09-04:
Any updates on Electra format? It uses 8 bytes instead of 5. In Romania almost all Intercoms are from Electra.

closed 2026-02-03 by GameTec-live