RfidResearchGroup/ChameleonUltra #212: [idea]

shayya, 2024-04-24:
When will HID PROX LF support please be added? 
I have 6 different lf fobs, and they're all hid prox. You'd think that would be the first lf protocol to be supported

GameTec-live, 2024-04-24:
as soon as someone finds time and wants to implement it... (this could be you or anyone else)

GameTec-live, 2024-04-29:
> This seems like a highly demanded feature

Yeah, EVERY somewhat common card that the chameleon could do is in high demand, but people are busy and RRG does god knows what (atleast there arent any recent commits from them). They (Most of the contributors, exception being xianglin1998) are doing that in their freetime and dont get paid...

> As far as I'm aware HID Prox is the the most used LF protocol in the united states

Maybe, but the US is only part of the world XD (So I personallly very rarely encounter LF cards, so i dont care too much about that)

shayya, 2024-04-29:
Interesting.. where I live about half the systems I encounter are LF (hid prox) and the rest are a mix of mifare and desfire

Michael-A-Sloan, 2024-09-26:
would love for someone to work on this. I would work on it if I knew how too lol

GameTec-live, 2024-09-26:
> would love for someone to work on this. I would work on it if I knew how too lol
> 

Get a devkit, HID prox card, A SWD debugger and a pc. Plug them together, set up the nrf build environment and start doing stuff...

Michael-A-Sloan, 2024-09-26:
> > would love for someone to work on this. I would work on it if I knew how too lol
> > 
> 
> Get a devkit, HID prox card, A SWD debugger and a pc. Plug them together, set up the nrf build environment and start doing stuff...

Since you so knowledgeable what is the stuff part? If I am to spend the kind of money to buy another system I should probably know that part don't you think?

GameTec-live, 2024-09-26:
"stuff" is whatever you want to work on lol. HID prox id assume as your writing on this issue.
You could also solder the SWD wires to the normal CU, but unlike the devkit it doesnt have headers.

tri4ng1e, 2025-06-24:
> Can CU read any 125khz cards

CU can read and emulate only one type of LF tag as of now (plain em410x). I hope some day we will be able to read and emulate other types, like HID Prox and others.

An-git-hub, 2025-07-01:
> read HID Prox II cards with ChameleonUltra

Where can I download your dfu-app.zip so that I could upload it to Chameleon Ultra?

jamisonderek, 2025-07-18:
@TeCHiScy Great Job!!! Do you want any help? You can reach out to me on Discord at @CodeAllNight.  I've added LF protocols to Chameleon Ultra (on the firmware/CLI side) so I know that part of the puzzle. I haven't looked at the supporting apps yet.

I'd love to see how you are doing the FSK demodulation (I checked your fork but didn't see any commits). I just finished implementing [Viking protocol](https://github.com/jamisonderek/ChameleonUltra/tree/jamisonderek/viking) which is using Manchester, RF/32 but I don't see how to do FSK2a modulation since `register_rio_callback(GPIO_INT0_callback);` only gives falling edge & we need to know every 8 or 10 cycles to demod.

I can see you are clearly getting Wiegand bits, facility and card number -- so you obviously figured it out. I've made [videos](https://www.youtube.com/playlist?list=PLM1cyTMe-PYKJeHWkN8vKntEyqmqlj_kY) on various LF protocols, but I need help understanding how to get the raw data for FSK2a (I understand how to demod if I know a bunch of FC/8, FC/10 values).

TeCHiScy, 2025-07-19:
> [@TeCHiScy](https://github.com/TeCHiScy) Great Job!!! Do you want any help? You can reach out to me on Discord at [@codeallnight](https://github.com/codeallnight) (note: I'm MrDerekJamison on YouTube and JamisonDerek on GitHub). I've added LF protocols to Chameleon Ultra (on the firmware/CLI side) so I know that part of the puzzle. I haven't looked at the supporting apps yet.
> 
> I'd love to see how you are doing the FSK demodulation (I checked your fork but didn't see any commits). I just finished implementing [Viking protocol](https://github.com/jamisonderek/ChameleonUltra/tree/jamisonderek/viking) which is using Manchester, RF/32 but I don't see how to do FSK2a modulation since `register_rio_callback(GPIO_INT0_callback);` only gives falling edge & we need to know every 8 or 10 cycles to demod.
> 
> I can see you are clearly getting Wiegand bits, facility and card number -- so you obviously figured it out. I've made [videos](https://www.youtube.com/playlist?list=PLM1cyTMe-PYKJeHWkN8vKntEyqmqlj_kY) on various LF protocols, but I need help understanding how to get the raw data for FSK2a (I understand how to demod if I know a bunch of FC/8, FC/10 values).

Hey, I would reach you out on Discord some time later.

PR #267 (2025-08-04): feat: add lf HIDProx read, t55xx write, emulate function
code quaility:
- consistance: simulation -> emulation, label -> tag
- machine translated unreadable comments are changed to native human readable english
- convert legacy `nrf_drv_xxx` macro usage to new `nrfx_xxx` macro usage
- make `byte_mirror` a common util, and share its definitions between hf & lf

fix:
- fixed a issue that cause em410x write args not verified in python cli

logic:
- newly added cli command includes: `lf hid prox read`, `lf hid prox write`, `lf hid prox econfig`. `hw slot list` is also updated to show HIDProx type  [... cut]
commit 098e0a91 (2025-08-04): feat: add lf HIDProx read, t55xx write, emulate function (#267)

* feat: add lf HIDProx read, t55xx write, emulate function

code quaility:
- consistance: simulation -> emulation, label -> tag
- machine translated unreadable comments are made native

logic:
- newly added cli command includes: `lf hid prox read`, `lf hid prox write`, `lf hid prox econfig`
- machester demodulator is simplified
- various wiegand formats of HIDProx are supported
- goertzel algorithm is used in  HIDProx FSK demod
- lf read is refactored using stream/feed pattern to boost scan speed
- t55xx write is refactored to sh [... cut]
closed 2025-08-04 by GameTec-live