RfidResearchGroup/ChameleonUltra #213: Chameleon Lite: NRF52840 HW issue. Need to reflash a new chip.

spp2000, 2024-05-18:
Hi guys,
I had an hardware issue with the NRF52840 chip. No bluetooth and no USB communication at all, while the emulation of the stored slots was still working.
Reflowing the chip with hot air didn't solve the problem, so I desoldered it. I found 3 broken pads under the chip (root cause?): the chip pads remained on the PCB.

[image]

(the aluminum foil is to protect the surrounding components during hot air operations)

I ordered a "ProMicro NRF52840" developer board as a donor board

[image]

I have successfully removed the chip and soldered on the Chameleon PCB (not an excellent job, but it should work).

[image]
[image]

Now I need help to flash it with this J-Link device: I mean both hardware connection and software procedure.

[image]

Thanks in advance

GameTec-live, 2024-05-18:
I told you that RMA would be the easier option Lol
But ok, anyways:
You want to get the hex file of the full firmware image (its under lite-firmware under github actions) to flash it, the easiest option would probably be openocd. So youd need to install openocd and then tell openocd to run this flash script (chuck it into a .ocd file, running openocd with -f  and specifying the file should run it):
```ocd
source [find interface/cmsis-dap.cfg]
source [find target/nrf52.cfg]

flash init; init; reset halt; flash erase_sector 0 1 last;
program fullimage-lite.hex verify
reset
exit
```

The line `source [find interface/cmsis-dap.cfg]` configures it for a raspberry pi pico debug probe, but you should be able to change it to `source [find interface/jlink.cfg]` (atleast from the bit of googling i did, might be a different name)

Then if you look at the back of the PCB, the SWD pads should be labled... Youll need to solder wires to SWD and SWC and GND. Then power up the Chameleon, your JLINK and flash it with openocd...

spp2000, 2024-05-18:
ahahah, but RMA is not so funny! ;)
I try to do what you wrote. Thanks

spp2000, 2024-05-19:
On the back I can see VCCNRF (that should be connected to J-Link VTref, right?), GND, and two pads: C and D

[image]

Following the traces and looking at the datasheet, I guess that C could be SWDIO (J-Link pin 7) and D could be SWDCLK (J-Link pin 9)

[image]

[image]

Can you confirm that the connections are right?

GameTec-live, 2024-05-19:
You do not connect any voltage anwhere. Just GND and the data pins... The chameleon ahs its own power circuit.
Them being labled C and D (Clock and Data) could be possible, so ig just try it and if it doesnt work, swap the maround...

spp2000, 2024-05-19:
C Clock, D Data...Thanks.
OK I'll try with only C,D and GND.
Btw VTref in not a power supply, it should be the pin used by J-Link to *read* the chip supply voltage (reference)

GameTec-live, 2024-05-19:
Ah, ok... Ig connecting it isnt a bad idea then...
Im not rich enough to have a J-link, im using the 15€ or so pi pico debug probe usually lmao