RfidResearchGroup/ChameleonUltra #214: Chameleon Ultra not able to emulate em410x cards

wintersushi, 2024-05-21:
Recently got a myself a Chameleon Ultra, using the Gui i was able to emulate, write and copy HF cards. Although, all of the LF rfid cards i have tried to emulate do not seem to work. I've tried a handful of different cards, as well as updating the firmware and clearing the slots. 

I am a beginner with all this but even when reading through the forums and watching videos, everyone seems to simply scan the card with the LF scanner, copy the UID and it works. I'm wondering if i'm missing something or if anyone has experienced anything similar and could help resolving this issue?

GameTec-live, 2024-05-21:
https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/troubleshooting.md

Or you might be affected by #175

wintersushi, 2024-05-21:
> https://github.com/RfidResearchGroup/ChameleonUltra/blob/main/docs/troubleshooting.md
> 
> 
> 
> Or you might be affected by #175 

I'm a little confused, if I can read the LF cards and copy it onto the Chameleon - how can it have glue blocking it and interfering with the emulation?

GameTec-live, 2024-05-21:
Ah, you can read the cards... Then there being glue doesnt really make sense...

wintersushi, 2024-05-21:
> Ah, you can read the cards... Then there being glue doesnt really make sense...

Yeah! I can read the LF cards and get the UID with no issue but whenever I try to emulate them with the chameleon, the blue and white light will turn on while next to the sensor but nothing happens. 

Only in one instance did it miraculously work, I tried hovering it over the sensor in different positions and different angles of the chameleon - maybe 1/10 times does it work for that one card. 

So I'm not sure if this is stupid question, and I'm suppose to have more than the UID or steps I've missed when copying the card?

GameTec-live, 2024-05-21:
Nope, that seems to be correct... probably a modulation issue or similar like in #175 then...

Nextery, 2024-05-24:
Ah lucky one. I can read LF cards only on 1.0 firmware, but UID is always 00000000. On 2.0 I’m getting an error “No card found, try to move Chameleon on card”

kayantoast, 2024-07-02:
> Ah lucky one. I can read LF cards only on 1.0 firmware, but UID is always 00000000. On 2.0 I’m getting an error “No card found, try to move Chameleon on card”

I just got the Chameleon ultra as well an am facing this exact same issue. Any solutions for this so far?

whywilson, 2024-07-03:
> > Ah lucky one. I can read LF cards only on 1.0 firmware, but UID is always 00000000. On 2.0 I’m getting an error “No card found, try to move Chameleon on card”
> 
> I just got the Chameleon ultra as well an am facing this exact same issue. Any solutions for this so far?

Contact your seller and get the replacement of the bottom board. It will solve the issues.

ffcrus, 2024-08-07:
> Contact your seller and get the replacement of the bottom board. It will solve the issues.
I have exactly the same problem. My device does not emulate LF cards. I ordered from indiegogo about a year ago. How can I contact the seller for the bottom board replacement?

Aleks1958, 2025-04-15:
Здравствуйте вопрос: Простой 125кГц еммарин к домафону элтис Хамемелеон реагирует так: загарается белый диод с боку и показывает синим слот в котором ключ, если держать то так и будет светить белым но не открывает, что еммарин уверен точно, делал Дубликатором ключей домофона и магнитных карт TMRF-2RW им делал таблетки обычные перезаписываемые к этому домафону элтис, таблетка открывает, её писал в хамемелеон но он открывать отказывается, подскажите что может быть кто знает? заранее благодарен.

brianbgarber, 2025-06-16:
> > Contact your seller and get the replacement of the bottom board. It will solve the issues.
> 
> I have exactly the same problem. My device does not emulate LF cards. I ordered from indiegogo about a year ago. How can I contact the seller for the bottom board replacement?

I'm in the same boat as I ordered from Indiegogo ... in April 2023. Did you find an answer to getting a new bottom board?

Aleks1958, 2025-06-16:
I just changed the device and it worked, what was the problem, I didn't
figure it out, всем добра